ricardoquesada/bluepad32 #107: [Bug]: i can't use a A2DP

ErickRBTK, 2024-06-14:
### What happened?

A bug happened! Hello, someone has tried to enable A2DP to send audio (if possible to the gamepad), simultaneously with the gamepad to send it to wireless headphones.

use the example BT_MUSIC_SENDER_WRITE from the ESP32-A2DP library 
I tried, but I get an error to compile

### Bluepad32 Version

4.0-beta2

### Bluepad32 version custom

Example: Using Git develop branch commit hash #xxxxxxx

### Bluepad32 Platform

Arduino IDE

### Platform version

E.g: Arduino IDE 1.8.19, Bluepad v4.0.4

### Controller

PS4

### Microcontroller

ESP32

### Microcontroller board

DOIT ESP32 DEVKIT V1 (ESP-WROOM-32)

### OS

Windows

### Relevant log output

```shell
In file included from C:\Users\ErickDaniel\Documents\Arduino\libraries\ESP32-A2DP-main\src/BluetoothA2DPSource.h:22,
                 from C:\Users\ErickDaniel\Desktop\ESP32 BT Controllers\tests\z_test\bt_music_sender\bt_music_sender.ino:17:
C:\Users\ErickDaniel\Documents\Arduino\libraries\ESP32-A2DP-main\src/BluetoothA2DPCommon.h:54:10: fatal error: esp_bt_main.h: No such file or directory
 #include "esp_bt_main.h"
          ^~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compilando para la tarjeta DOIT ESP32 DEVKIT V1.
```

### Relevant sketch

```shell
#include "BluetoothA2DPSource.h"
#include <math.h> 

#define c3_frequency  130.81

BluetoothA2DPSource a2dp_source;

// The supported audio codec in ESP32 A2DP is SBC. SBC audio stream is encoded
// from PCM data normally formatted as 44.1kHz sampling rate, two-channel 16-bit sample data
int32_t get_data_frames(Frame *frame, int32_t frame_count) {
    static float m_time = 0.0;
    float m_amplitude = 10000.0;  // -32,768 to 32,767
    float m_deltaTime = 1.0 / 44100.0;
    float m_phase = 0.0;
    float pi_2 = PI * 2.0;
    // fill the channel data
    for (int sample = 0; sample < frame_count; ++sample) {
        float angle = pi_2 * c3_frequency * m_time + m_phase;
        frame[sample].channel1 = m_amplitude * sin(angle);
        frame[sample].channel2 = frame[sample].channel1;
        m_time += m_deltaTime;
    }
    // to prevent watchdog
    delay(1);

    return frame_count;
}

void setup() {
  a2dp_source.set_auto_reconnect(false);
  a2dp_source.start("F-SL001A", get_data_frames);  
  a2dp_source.set_volume(60);
}

void loop() {
  // to prevent watchdog in release > 1.0.6
  delay(3000);
  a2dp_source.start("F-SL001A", get_data_frames);  
}
```

ricardoquesada, 2024-06-15:
Bluepad32 uses BTstack as the Bluetooth stack, and not the one from esp-idf.
So,  if you want to use A2DP, you have to use the BTstack API for that.

Search in the BTstack examples, there might be one:
https://github.com/bluekitchen/btstack/tree/master/example

closed 2024-06-15 by ricardoquesada

ErickRBTK, 2024-06-15:
Thank you very much, this gives me a clue to see if I can integrate it into my project or look for another alternative.