ricardoquesada/bluepad32 #108: Cannot connect ESP32S3 DevKitM-1 to PS4 Controller

sagar-bei031, 2024-06-16:


sagar-bei031, 2024-06-16:
Using IDF
Output from idf.py monitor:

Could not find property 11
BTstack up and running at F4:12:FA:88:13:FA
HCI not ready, cannot send packet, will again try later. Current state idx=1
custom: on_init_complete()
BLE scan -> 1
custom: Bluetooth enabled: 1
Deleting stored BLE link keys:
                                                                                                                                                                                  .
Failed command: HCI_EVENT_COMMAND_COMPLETE: opcode = 0x0c05 - status=1

Using Arduino IDE
Output from Arduino Controller:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
Bluepad32 (C) 2016-2024 Ricardo Quesada and contributors.
Version: v4.1.0
BTstack: Copyright (C) 2017 BlueKitchen GmbH.
BTstack up and running at F4:12:FA:88<1Firmware: Bluepad32 for Arduino v4.1.0
BD Addr: F4:12:FA:88:13:F9

PS4 does connect after holding share and PS button.

ricardoquesada, 2024-06-17:
ESP32-S3 does not support Bluetooth Classic, the protocol used by DualShock 4.

https://bluepad32.readthedocs.io/en/latest/supported_gamepads/#bluetooth-gamepads-and-their-protocol

closed 2024-06-17 by ricardoquesada