ricardoquesada/bluepad32 #116: CAPS lock LED Report

peterburk, 2024-08-25:
**Is your feature request related to a problem? Please describe.**
I'm very pleased with the bluepad32 examples, and successfully used it on a RPi Pico W to make a Bluetooth - USB HID keyboard bridge this weekend! However, the CAPS lock LED doesn't come on. 

**Describe the solution you'd like**
Update the uni_bt_service.gatt to include ORG_BLUETOOTH_CHARACTERISTIC_REPORT_MAP as seen in the btstack example hog_keyboard_demo.c. 

**Describe alternatives you've considered**
I've looked into patching uni_bt_service myself, but would prefer for this change to be made upstream so it can benefit more users. 

**Additional context**
I'm using an Apple Magic Keyboard (which doesn't work with the HID remapper project, but does work with bluepad32!).

ricardoquesada, 2024-08-25:
if you could share the code that you did, I can make the needed changes (if needed) and apply them. ty

peterburk, 2024-08-28:
I see that there's already uni_hid_parser_keyboard_set_leds in uni_hid_parser_keyboard.c - thank you for implementing that! That's much cleaner than my hacks to uni_bt_service. 

Unfortunately I haven't got it to work, due to the error "Keyboard: Set LED report not implemented for BR/EDR yet". 

I always get status = ERROR_CODE_UNKNOWN_CONNECTION_IDENTIFIER 0x02. 

When I try to create a new connection using hid_host_create_connection, it still fails due to !connection->control_cid. 

Is there any plan to implement LED reports for BR/EDR (Classic Bluetooth)?

ricardoquesada, 2024-09-01:
Correct, for BR/EDR keyboards I'd need to know which report Id I should should... perhaps report Id 1 should work, I can give it a try

LeeNX, 2025-01-06:
Would https://github.com/Mystfit/ESP32-BLE-CompositeHID/blob/master/examples/KeyboardExamples/KeyboardLEDs/KeyboardLEDs.ino be helpful in debugging and testing with both host and client devices having and accessible stack?