ricardoquesada/bluepad32 #121: [Bug]: Issues with L298N

kleinKN23, 2024-09-13:
### What happened?

Been trying to use my Xbox One controller to control an rc car, but when i create an object for the motor, the controller doesnt want to connect anymore. Works fine with example file, but when i add this line sepcifically stops working :
// Create one motor instance
L298N motor(EN, IN1, IN2);

### Bluepad32 Version

I don't know

### Bluepad32 version custom

Example: Using Git develop branch commit hash #xxxxxxx

### Bluepad32 Platform

Other

### Platform version

Arduino IDE 2.3.2

### Controller

Xbox One, 1708

### Microcontroller

ESP32

### Microcontroller board

NodeMCU esp32S

### OS

None

### Relevant sketch

```shell
// Copyright 2021 - 2023, Ricardo Quesada
// SPDX-License-Identifier: Apache 2.0 or LGPL-2.1-or-later

/*
 * This example shows how to use the Controller API.
 *
 * Supported on boards with NINA W10x. In particular, these boards:
 *  - Arduino MKR WiFi 1010,
 *  - UNO WiFi Rev.2,
 *  - Nano RP2040 Connect,
 *  - Nano 33 IoT,
 *  - Arduino MKR Vidor 4000
 */
#include <Bluepad32.h>
#include <L298N.h>

// Pin definition
const unsigned int IN1 = 7;
const unsigned int IN2 = 8;
const unsigned int EN = 9;

// Create one motor instance
L298N motor(EN, IN1, IN2);
ControllerPtr myControllers[BP32_MAX_CONTROLLERS];

// Arduino setup function. Runs in CPU 1
void setup() {
  // Initialize serial
  Serial.begin(115200);
[... 199 lines cut ...]
// Arduino loop function. Runs in CPU 1
void loop() {
  // This call fetches all the controller info from the NINA (ESP32) module.
  // Call this function in your main loop.
  // The controllers' pointer (the ones received in the callbacks) gets updated
  // automatically.
  BP32.update();

  // It is safe to always do this before using the controller API.
  // This guarantees that the controller is valid and connected.
  for (int i = 0; i < BP32_MAX_CONTROLLERS; i++) {
    ControllerPtr myController = myControllers[i];

    if (myController && myController->isConnected()) {
      if (myController->isGamepad())
        processGamepad(myController);
      else if (myController->isMouse())
        processMouse(myController);
      else if (myController->isBalanceBoard())
        processBalanceBoard(myController);
    }
  }
  delay(150);
}
```

raffitaff, 2024-09-14:
Not sure if anyone can respond for T/S.

The pins you use (7,8,9) might be worth taking a look at to see if they can be used as GPIO. [I think this is your board,](https://www.waveshare.com/nodemcu-32s.html) but double check. If so, Pin 9 is connected to flash memory and those (pins 7-9) look to also be used on startup, so you have to take care of their states when flashing if you are using them. 

If that is the pinout of your board, I would test this issue using other available pins.

kleinKN23, 2024-09-14:
Thanks for your answer!

You might be right! Just tested it with pins 35, 32 and 33 and it
connected. Seems like I havent done my debugging right.

Thanks again for your answer!

closed 2024-09-14 by kleinKN23