ricardoquesada/bluepad32 #122: Conexión a un solo control

NeftaliVenancio, 2024-09-17:
¿Es posible declarar en el codigo de arduino las caracteristicas de un unico control al que se deba conectar (Control de xbox one)?

ricardoquesada, 2024-09-17:
https://bluepad32.readthedocs.io/en/latest/FAQ/#how-to-pair-just-one-controller-to-one-particular-board

closed 2024-09-17 by ricardoquesada

NeftaliVenancio, 2024-09-19:
Muchas gracias.

Si entiendo bien, con esto declaro la direción del controlador permitido, pero a la hora de que busque coenctarse, el codigo sigue buscando el tipo de control dentro de la lista de posibilidades o en automatico busca por ejemplo "Xbox one" si es el caso.

ricardoquesada, 2024-09-20:
solamente los controladores que tengan la "BT address" (a veces llamado Mac address) van a poder conectarse con el ESP32.

NeftaliVenancio, 2024-09-20:
Sí, entiendo. Pero adicional puedo indicarle por defecto que tipo de control es la BTaddress? Lo que pasa es que en este moemnto lo uso de modo que recuerde el dispositivo que esta vinculado, pero en ocasioens me lo reconoce como codigo 30 ("Xbox") y y no 31("Xbox one") que es el correcto, esto causa que n oreconozca los botones corectamente. hay forma de decirle que tipo de control es de forma estatica?