ricardoquesada/bluepad32 #124: [Bug]: 8BitDo Pro 2 loses connection after 30s

jgrgt, 2024-09-23:
### What happened?

I'm using this sketch: https://github.com/JohnFodero/MiniSkidi-V3.0/tree/main/MiniSkidi_3.0
With version 4.1.0 and my 8BitDo Pro 2. My controller loses connection every 30s. The controller shuts down after this (it seems...) 

Typically I would get something like this:
```
...
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    0,    0, axis R:  511,   11, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0, battery: 0
CALLBACK: Controller is disconnected from index=0
```

I did get 1 crash so far, I'm not sure if it's related, but just in case, here's the log:

```
idx=0, dpad: 0x00, buttons: 0x0000, axis L:   -8,  511, axis R:    0,  511, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0, battery: 0

idx=0, dpad: 0x00, buttons: 0x0000, axis L:   -8,  511, axis R:    0,  511, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0, battery: 0

idx=0, dpad: 0x00, buttons: 0x0000, axis L:   -8,  511, axis R:    0,  511, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0, battery: 0

idx=0, dpad: 0x00, buttons: 0x0000, axis L:   -8,  511, axis R:    0,  511, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0, battery: 0

Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:

PC      : 0x00000000  PS      : 0x00060230  A0      : 0x801061b9  A1      : 0x3ffbc270  

A2      : 0x0000d326  A3      : 0x3ffc5c1c  A4      : 0x3ffd3ee4  A5      : 0x00000000  

A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800f8c3c  A9      : 0x3ffbc250  

A10     : 0x3ffc5c1c  A11     : 0x0000d326  A12     : 0x3ffc3cf8  A13     : 0x00060223  

A14     : 0x00060220  A15     : 0x00000001  SAR     : 0x00000017  EXCCAUSE: 0x00000014  

EXCVADDR: 0x00000000  LBEG    : 0x400e5e00  LEND    : 0x400e5e14  LCOUNT  : 0x00000000  

Backtrace: 0xfffffffd:0x3ffbc270 0x401061b6:0x3ffbc290 0x400f8d8e:0x3ffbc2c0 0x400e45c3:0x3ffbc2e0 0x401652c7:0x3ffbc300

ELF file SHA256: ebf84de85360f6a9

Rebooting...

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0030,len:1344

load:0x40078000,len:13964

load:0x40080400,len:3600

entry 0x400805f0

Bluepad32 (C) 2016-2024 Ricardo Quesada and contributors.

Version: v4.1.0

BTstack: Copyright (C) 2017 BlueKitchen GmbH.

BTstack up and running at EC:62:60:9D:52:AE

1

1

Firmware version installed: Bluepad32 for Arduino v4.1.0

SKIDDY TIME

BD Address: EC:62:60:9D:52:AE
```

The only thing I've tried so far is to comment out line 138: `BP32.forgetBluetoothKeys();`. But this did not improve anything.

### Bluepad32 Version

Other

### Bluepad32 version custom

4.1.0

### Bluepad32 Platform

Arduino IDE

### Platform version

Arduini IDE 2.3.2

### Controller

8BitDo Pro 2

### Microcontroller

ESP32

### Microcontroller board

ESP32 Dev Module

### OS

macOS

ricardoquesada, 2024-09-25:
does it crash with the unmodified example ?

If no, then probably it crashes due to the additions in the code.
If yes, then... mmm... weird... perhaps a faulty esp32 (?)

jgrgt, 2024-09-25:
Thx for your reply @ricardoquesada. I've experimented a bit and it does happen for the 'controller' example too, on that ESP32 and another one (different type). But then I tried switching the Pro 2 to 'Android' mode with the selection switch on the back and the problem was gone. It was in 'Xbox (One)' mode during all my previous tests.

So the bug seems to be only for that mode. But the workaround is easy: just switch to Android mode.