ricardoquesada/bluepad32 #127: [Bug]: Dualshock 4 (clone) touchpad and LED not supported?

dwilliss, 2024-10-11:
### What happened?

I don't know if this is a bug, or a not-completely implemented feature or if this Dualshock 4 knock-off is just garbage.  Probably the latter, but thought I'd ask before I spend money on an official controller.

I can get responses back from everything on the thing except the touch pad.  Looking at the code, it looks like it should give me `deltaX()` and `deltaY()` values as well as a button values when clicked, but it doesn't respond.
Also, the `setColorLED` doesn't seem to do anything.

I'm using Bluepad32 version 4.1.0 which is what it gave me when I dowloaded it from the Arduino IDE yesterday.
I don't know if this is meaningful, but the controller reports VendorID 0x054c, Product ID 0x09cc, flags = 0x0005

I also dug through the parser code for ds4 to see why the 2 buttons on the bottom of the controller weren't supported.  They're M1 and M2.  It turns out they're macro buttons that you can assign to combinations of other buttons.

I have to say though that this package is better than the PS4Controller Arduino library which refused to pair at all, even after following their convoluted instructions to manually set the bluetooth address of the ESP32 on the controller.  Yours just pairs the way you'd expect.

### Bluepad32 Version

Other

### Bluepad32 version custom

Version 4.1.0 which isn't on your dropdown list.
This is the version the Arduino IDE gave me when I installed it.

### Bluepad32 Platform

Arduino IDE

### Platform version

Version: 2.3.3
Date: 2024-09-25T09:41:18.242Z
CLI Version: 1.0.4

### Controller

A Dualshock 4 clone purchased off [Amazon](https://www.amazon.com/dp/B0C1VMH1T8?th=1)

### Microcontroller

ESP32

### Microcontroller board

[Adafruit ESP32 Feather V2](https://learn.adafruit.com/adafruit-esp32-feather-v2/overview)

### OS

Windows

### Relevant sketch

```shell
// Slightly modified version of the sample program.  Changed the dumpGamepad function
// to show battery(), deltaX() and deltaY() and to try to cycle the LED color through
// red, green and blue every second.

#include <Bluepad32.h>

ControllerPtr myControllers[BP32_MAX_GAMEPADS];

// This callback gets called any time a new gamepad is connected.
// Up to 4 gamepads can be connected at the same time.
void onConnectedController(ControllerPtr ctl) {
    bool foundEmptySlot = false;
    for (int i = 0; i < BP32_MAX_GAMEPADS; i++) {
        if (myControllers[i] == nullptr) {
            Serial.printf("CALLBACK: Controller is connected, index=%d\n", i);
            // Additionally, you can get certain gamepad properties like:
            // Model, VID, PID, BTAddr, flags, etc.
            ControllerProperties properties = ctl->getProperties();
[... 269 lines cut ...]
    // When enabled, controllers like DualSense and DualShock4 generate two connected devices:
    // - First one: the gamepad
    // - Second one, which is a "virtual device", is a mouse.
    // By default, it is disabled.
    BP32.enableVirtualDevice(false);
}

// Arduino loop function. Runs in CPU 1.
void loop() {
    // This call fetches all the controllers' data.
    // Call this function in your main loop.
    bool dataUpdated = BP32.update();
    if (dataUpdated)
        processControllers();

    // The main loop must have some kind of "yield to lower priority task" event.
    // Otherwise, the watchdog will get triggered.
    // If your main loop doesn't have one, just add a simple `vTaskDelay(1)`.
    // Detailed info here:
    // https://stackoverflow.com/questions/66278271/task-watchdog-got-triggered-the-tasks-did-not-reset-the-watchdog-in-time

    //     vTaskDelay(1);
    delay(150);
}
```

ricardoquesada, 2024-10-11:
mmm...

LED and touch are supported in DualShock4.

And some clones support those features, but others don't.

I think you might have one of those clones that don't support LED and/or touchpad.

But you can try connecting the controller to a Linux (or Windows) machine via Bluetooth. Can you control the mouse from the touchpad ?

dwilliss, 2024-10-11:
I just tried that (Windows).  Clicking the "home" button on the controller brings up the Steam client.
Windows doesn't see it as a mouse, but the Steam client has a "Test Device Inputs" option.
That seems to see the touchpad just fine.
There's also an option to change the LED color which does nothing.  So that must not be implemented on this one.

If bluepad32 was implemented as an Arduino library, I know I could go into the library folder and insert some debugging print statements to see what's going on.  I'm not sure how to do that with this though because it seems to be installed as pre-compiled binaries in the Arduino IDE and I don't know how to build them.
If one of your doc files tells how to do that, I can try messing with it this weekend and see what actually comes back.

ricardoquesada, 2024-10-11:
yes, I thought of decoupling bluepad32 for the "esp32 + bluepad32 boards" for this scenario. I might  do it some day.

In the meantime you can play with the esp32 example (requires that you install esp-idf) and you can modify bluepad32 code.

https://github.com/ricardoquesada/bluepad32/tree/main/examples/esp32

ricardoquesada, 2024-10-11:
alternative, I might be able to look at the controller. Here are the instructions:

https://bluepad32.readthedocs.io/en/latest/FAQ/#could-you-add-support-for-my-unsupported-controller