ricardoquesada/bluepad32 #128: ESP32 custom Bluetooth MAC address

shurillu, 2024-11-15:
Hello Ricardo,
   as agreed in the discord chat, I'll explain my request.

Use case
Manage/drive several device with the same controller (obviously one at time), without re-pair the controller with the device.
Example: I have several combat antbot (something like battlebots/robotwars bot, but smaller and made of 3D printed parts) and I use ESP32 + bluetooth joypads to drive it. Setting the same bluetooth address in every bots makes the pairing process simpler (turn on the bot, turn on the joypad without the need of re-pair process).
Another advantage is: you have already paired the controller with the device, so there is no risk that another device could "catch" your controller during the pairing phase (think a competition with several bot belonging to different owners, you can't go to every guy saying "please please turn off all your bot because I have to re-pair mine!!").

Possible solution
There are two function on ESP-IDF SDK:

`esp_err_t esp_read_mac(uint8_t mac, esp_mac_type_t type)` [doc here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/misc_system_api.html#_CPPv412esp_read_macP7uint8_t14esp_mac_type_t)
`esp_err_t esp_base_mac_addr_set(uint8_t* mac)` [doc here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/misc_system_api.html#_CPPv421esp_base_mac_addr_setPK7uint8_t)

The first function retrieve the current MAC address, the second function set the new MAC address.
I use these function in my library for DualShock4 controller.

I tried to modify the bt address, but your library seems to save it somewhere (maybe because already initialized) so the changes doesn't affect the address used by your library.
Here what I did:

1. Set the Arduino environment to use your library
2. Loaded your Arduino example (Controller.ino)
3. in the `setup()`, at the beginning and just below the serial initialization I added this snippet (to set the bluetooth MAC address)
```
    // Define your new MAC address
    uint8_t newMAC[] = { 0x08, 0x03, 0x19, 0x77, 0x00, 0x00 - 0x02 };

    // Set the new MAC address
    if (esp_base_mac_addr_set(newMAC) == ESP_OK) {
        Serial.println("\n\nMAC address set successfully");
    }
    else {
        Serial.println("\n\nFailed to set MAC address");
    }
```
4. Just before the end of the setup() block, I added this snippet (to print the bluetooth MAC address)
```
    esp_read_mac(baseMac, ESP_MAC_BT);
    Serial.print("Bluetooth MAC: ");
    for (int i = 0; i < 5; i++) {
        Serial.printf("%02X:", baseMac[i]);
    }
    Serial.printf("%02X\n", baseMac[5]);
```
Compiling and uploading, this is what I get (Serial monitor):
```
Bluepad32 (C) 2016-2024 Ricardo Quesada and contributors.
Version: v4.1.0
BTstack: Copyright (C) 2017 BlueKitchen GmbH.
BTstack up and running at 48:F3:52:AD:C3:76
VMDPR_

MAC address set successfully

Firmware: Bluepad32 for Arduino v4.1.0
BD Addr: 48:F3:52:AD:C3:76
Bluetooth MAC: 08:03:19:77:00:00
```
In other word, your library start with the default ESP32 bluetooth MAC address, then I change the MAC address successfully, then I call your library member function `BP32.localBdAddress()` and the output is the default bt address (as nothing was done) but calling the ESP-IDF SDK function `esp_read_mac()` returns the custom bt address set before.

A proof that your library is still using the default bt address can be retrieved by pairing a controller and than reading the bt address paired.
I did it with a DualShock 4 pad, after paired I used the sixaxispairtool to read the bt address stored in the controller.

So I guess that something must be done inside your library, maybe deinitializing the library, changing the bt address and then initializing once again the library.

Hoping that my explanation is clear enough, thank you for you huge work!

Regards,

Stefano

ricardoquesada, 2025-01-02:
yes, you are right about it.

I'd say that if you call `esp_base_mac_addr_set()`, then you should call `esp_read_mac()` to read it.

I could implement a "set mac address" API in Bluepad32, but I need to see what BTstack is doing.

It already has `custom_bd_addr` that some chipsets support it, but apparently it is not implement  in ESP32 (?).

I need to look into it.

https://github.com/bluekitchen/btstack/blob/565100c1cae73dc1c5f9b239953dbdd14ae283d3/src/hci.h#L1328

shurillu, 2025-01-14:
Hello Ricardo,
   thanks for the reply.
I'm not an BlueKitchen BTStack library, but I found that:
[Here](https://github.com/bluekitchen/btstack/blob/master/src/hci.c#L1872) seems to be an initializing function and just [below some lines of code](https://github.com/bluekitchen/btstack/blob/master/src/hci.c#L1928) there is a case for initializing a custom BT address. 
The problem could be the #ifndef [here](https://github.com/bluekitchen/btstack/blob/master/src/hci.c#L1904): if is defined somewhere `HAVE_HOST_CONTROLLER_API`, the bluetooth custom address setter will not be executed.

I don't know if it could helps.

Regards,

Stefano