ricardoquesada/bluepad32 #129: [Request for insight] Ideas on reproducible builds?

LeeNX, 2024-11-23:
Looking for some info or suggestions on how to make reproducible builds and/or firmware.  
  
I ran into a problem of late, where I can't seem to make a functioning ```bluepad32``` build, where a few weeks back (might be a little longer), that I am sure I had a working test bluepad32 on both ESP32-C3 and the original ESP32 board, connecting and function with other ESP32-C3 and original ESP32 boards using ESP32-BLE-Gamepad.  

I tried rolling back my Arduino libraries and board updates, but I don't think the roll backs are working, so I was thinking of trying the PlatformIO route, but don't see any board setup for ```bluepad32```.  
  
Does anybody have any suggestions I might be able to try?

ricardoquesada, 2024-11-25:
IIUC the client is using ESP32-BLE-Gamepad, and the "server" is using Bluepad32.

At one point it was working, but not anymore. And you want to have "builds"... is that correct ?

If so, I used to have daily builds...and I think they are still there, at least for Blueapd32... but I guess that won't help you.
How are you compiling Bluepad32? what project are you using?the "template" ?

LeeNX, 2024-11-25:
Using ESP32-BLE-Gamepad as the client and Bluepad32 as the server is the end goal and I got it working.  
  
Embarrassingly, I had two problems, first been a charging only USB-C cable, which mean I could not program or monitor the ESP32-C3 Super Mini and the second problem, between the ESP32 original and C3 mini, their LED GPIO are different and would basically halt ESP32 firmware start just after the Bluepad32 version string.  
  
I did believe that I might have borked my local install, either the ESP32 release update or NimBLE update or something and then even downgrading the version of the libraries did not seem like it made any difference.  
  
Been playing with PlatformIO and the projects that can have locked version for dependancies, unlike the Arduino IDE.  
  
I did not find any reference to Bluepad32 supporting PlatformIO and was wonder how other members and possible yourself,  might be dealing with builds, that would have a good know state. As an example, the ESP32-BLE-Gamepad has an issue where users are not able to pair a second ESP32-BLE-Gamepad device. We all seem to be using the same sketch, but I have been wondering if it's a dependancy library bug, this might be an idea to rather make a firmware to share vs sketch.   
  
Thanks for the info and feedback.

ricardoquesada, 2025-01-01:
PlatformIO is working for me if you:

- remove the pio cache (e.g: `rm -rf ~/.platformio` and any other local reference).

It works for the "examples/esp32" and the "template project".

Try `develop` branch of this github project for instructions on PlatformIO

https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/tree/develop