ricardoquesada/bluepad32 #130: [Bug]: BP32.enableNewBluetoothConnections(false) does not appear to function

ste2425, 2024-11-25:
### What happened?

I setup  clean installation of the https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template using VSCode and PlatformIO on a system running Linux Mint 22 Cinnamon.

The only change that has been made to the template is to add `P32.enableNewBluetoothConnections(false)` at the end of `setup` . I have attached the full setup method to the code section below.

When i use a DS5 controller which has never been paired with the ESP32 before i would expect it to not pair. However it does indeed connect.

### Bluepad32 Version

Other

### Bluepad32 version custom

I don't know, i assume latest. I followed the instructions from the template here: https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template

### Bluepad32 Platform

Arduino ESP-IDF

### Platform version

E.g: Arduino IDE 2.1.0, or ESP-IDF v4.4.4, or CircuitPython 8.0.0, etc.

### Controller

Replicated with:

* Switch Pro
* DS4
* DS5

### Microcontroller

ESP32

### Microcontroller board

PlatformIO, Espressif32@5.4.0

### OS

Linux

### Relevant sketch

```shell
void setup() {
    Console.printf("Firmware: %s\n", BP32.firmwareVersion());
    const uint8_t* addr = BP32.localBdAddress();
    Console.printf("BD Addr: %2X:%2X:%2X:%2X:%2X:%2X\n", addr[0], addr[1], addr[2], addr[3], addr[4], addr[5]);

    // Setup the Bluepad32 callbacks
    BP32.setup(&onConnectedController, &onDisconnectedController);

    // "forgetBluetoothKeys()" should be called when the user performs
    // a "device factory reset", or similar.
    // Calling "forgetBluetoothKeys" in setup() just as an example.
    // Forgetting Bluetooth keys prevents "paired" gamepads to reconnect.
    // But it might also fix some connection / re-connection issues.
    BP32.forgetBluetoothKeys();

    // Enables mouse / touchpad support for gamepads that support them.
    // When enabled, controllers like DualSense and DualShock4 generate two connected devices:
    // - First one: the gamepad
    // - Second one, which is a "virtual device", is a mouse.
    // By default, it is disabled.
    BP32.enableVirtualDevice(false);

    // Enables the BLE Service in Bluepad32.
    // This service allows clients, like a mobile app, to setup and see the state of Bluepad32.
    // By default, it is disabled.
    BP32.enableBLEService(false);

    BP32.enableNewBluetoothConnections(false); // THIS BIT HAS NO EFFECT
}
```

ricardoquesada, 2024-11-25:
very weird.

And IIUC, it happens with any controller, correct ?

Which Bluepad32 version are you using ?

ste2425, 2024-11-26:
Yep it does it with every controller I have access to, DS4, DS5 and switch pro. 

I was sure to pair these controllers with the games console before the ESP so they were pairing fresh to the ESP if that makes sense. 

I will double check the BP32 version and report back. 

I also have access to the BP32 console, I will try disabling new connects via there and report back too.

ste2425, 2024-11-26:
Ok so i have done some more investigations and i think i have found the cause.

It appears to be a timing thing.

To my understanding following the code it is an async action, starting/stopping the service. We execute tasks on the BTStack core. 

In the platform init complete method `arduino_on_init_complete` it sets new connections to true.

What i think is happening (and by looking at the logs) by the time the startup method is called to disable new connections it is still in the process of enabling them.

So i believe its a race condition if that makes sense.

For people with IDF setup the solution is simple i think. Change the platform complete handler to remove the enabling of the service.

For people using the Arduino IDE it may not be so simple. Maybe The `BP32.setup` method could have an optional options object to set the default state for new connections?

Hope all that makes sense!

TLDR; It's a timing issue because the Arduino platform defaults to enabling new connections. If i do a non sync delay and then turn off new connections it works fine. Ideal solution would be to allow the platform to not enable new connections by default.

ricardoquesada, 2025-01-01:
mmm... this is an interesting bug.

It seems that only one "safe" function can be called per "loop".

in this case, `BP32.enableBLEService()` and `BP32.enableNewBluetoothConnections()` are competing for each other.

I'll  fix that....

ricardoquesada, 2025-01-01:
fixed.

closed 2025-01-01 by ricardoquesada

ricardoquesada, 2025-01-01:
Also moved "start scanning" to "setup time":

https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/commit/cd38b1234773ee2ea46c3e10fa4644631ba97993