ricardoquesada/bluepad32 #131: [Bug]: USB HID Report collison

controllercustom, 2024-12-06:
### What happened?

Compiler errors when using bluepad32 with ESP32 USB HID.

### Bluepad32 Version

Other

### Bluepad32 version custom

Arduino esp32_bluepad32 version 4.1.0

### Bluepad32 Platform

Arduino IDE

### Platform version

Arduino IDE 1.8.19 on Ubuntu Linux

### Controller

None

### Microcontroller

ESP32-S3

### Microcontroller board

Espressif ESP32-S3  DevKit-1 v 1.0

### OS

Linux

### Relevant log output

```shell
Arduino: 1.8.19 (Linux), Board: "ESP32S3 Dev Module, Disabled, Disabled, QIO 80MHz, 4MB (32Mb), Core 1, Core 1, USB-OTG (TinyUSB), Disabled, Disabled, Disabled, UART0 / Hardware CDC, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), 921600, None, Disabled"

In file included from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/libraries/USB/src/USBHID.h:22,
                 from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/libraries/USB/src/USBHIDMouse.h:23,
                 from /home/me/Sync/bluepad32-esp32ard/examples/usbhid/usbhid.ino:3:
/tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/tools/sdk/esp32s3/include/arduino_tinyusb/tinyusb/src/class/hid/hid.h:87:3: error: 'HID_REPORT_TYPE_INPUT' conflicts with a previous declaration
   HID_REPORT_TYPE_INPUT,      ///< Input
   ^~~~~~~~~~~~~~~~~~~~~
In file included from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/tools/sdk/esp32s3/include/btstack/src/btstack.h:62,
                 from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/tools/sdk/esp32s3/include/bluepad32/include/uni_hid_device.h:8,
                 from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/tools/sdk/esp32s3/include/bluepad32/include/platform/uni_platform.h:15,
                 from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/tools/sdk/esp32s3/include/bluepad32_arduino/include/arduino_platform.h:29,
                 from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/tools/sdk/esp32s3/include/bluepad32_arduino/include/ArduinoControllerData.h:9,
                 from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/tools/sdk/esp32s3/include/bluepad32_arduino/include/ArduinoController.h:13,
                 from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/tools/sdk/esp32s3/include/bluepad32_arduino/include/ArduinoBluepad32.h:12,
                 from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/tools/sdk/esp32s3/include/bluepad32_arduino/include/Bluepad32.h:7,
                 from /home/me/Sync/bluepad32-esp32ard/examples/usbhid/usbhid.ino:1:
/tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/tools/sdk/esp32s3/include/btstack/src/btstack_hid.h:103:5: note: previous declaration 'hid_report_type_t HID_REPORT_TYPE_INPUT'
     HID_REPORT_TYPE_INPUT,
     ^~~~~~~~~~~~~~~~~~~~~
In file included from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/libraries/USB/src/USBHID.h:22,
                 from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/libraries/USB/src/USBHIDMouse.h:23,
                 from /home/me/Sync/bluepad32-esp32ard/examples/usbhid/usbhid.ino:3:
/tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/tools/sdk/esp32s3/include/arduino_tinyusb/tinyusb/src/class/hid/hid.h:88:3: error: 'HID_REPORT_TYPE_OUTPUT' conflicts with a previous declaration
   HID_REPORT_TYPE_OUTPUT,     ///< Output
   ^~~~~~~~~~~~~~~~~~~~~~
In file included from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/tools/sdk/esp32s3/include/btstack/src/btstack.h:62,
                 from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/tools/sdk/esp32s3/include/bluepad32/include/uni_hid_device.h:8,
                 from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/tools/sdk/esp32s3/include/bluepad32/include/platform/uni_platform.h:15,
                 from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/tools/sdk/esp32s3/include/bluepad32_arduino/include/arduino_platform.h:29,
                 from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/tools/sdk/esp32s3/include/bluepad32_arduino/include/ArduinoControllerData.h:9,
[... 76 lines cut ...]
   ^~~~~~~~~~~~~~~~~~~~~
In file included from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/libraries/USB/src/USBHID.h:22,
                 from /tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/libraries/USB/src/USBHIDMouse.h:23,
                 from /home/me/Sync/bluepad32-esp32ard/examples/usbhid/usbhid.ino:3:
/tmp/autobuild_bluepad32ard_50439/arduino-1.8.19/portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/tools/sdk/esp32s3/include/arduino_tinyusb/tinyusb/src/class/hid/hid.h:1139:2: error: expected declaration before '}' token
  }
  ^
exit status 1
Error compiling for board ESP32S3 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

### Relevant sketch

```shell
#include <Bluepad32.h>
#include "USB.h"
#include "USBHIDMouse.h"

void setup() { }
void loop() { }
```

ricardoquesada, 2025-01-01:
weird.

if you try the template project ( https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template ) , `develop` branch, does ti work ?

Since it uses esp-idf 5.3.2 and Arduino Core v3.1.0

controllercustom, 2025-01-02:
> if you try the template project ( https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template ) , develop branch, does ti work ?

I tried the template project and set-target to esp32s3. I added some lines to sketch.cpp like this. 

```
#include <Arduino.h>
#include <Bluepad32.h>
#include <USB.h>
#include <USBHIDMouse.h>
USBHIDMouse Mouse;
...
void setup() {
    // initialize mouse control:
    Mouse.begin();
    USB.begin();
...    
```

I think TinyUSB is disabled in sdkconfig so no collision errors but it says USBHIDMouse does not name a type.

```
$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/me/Sync/bluepad32-esp-idf-arduino/esp/my_project/build
Executing "ninja all"...
[4/15] Performing build step for 'bootloader'
[1/1] cd /home/me/Sync/bluepad32-esp-idf-arduino/esp/my_project/build/bootloader/esp-idf/esptool_py && /home/me/.espressif/python_env/idf4.4_py3.10_env/bin/python /home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/me/Sync/bluepad32-esp-idf-arduino/esp/my_project/build/bootloader/bootloader.bin
Bootloader binary size 0x52b0 bytes. 0x2d50 bytes (35%) free.
[7/13] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/sketch.cpp.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/sketch.cpp.obj 
/home/me/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/my_project/build/config -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/my_project/main -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/newlib/platform_include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/freertos/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/freertos/include/esp_additions/freertos -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/freertos/port/xtensa/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/freertos/include/esp_additions -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/esp_hw_support/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/esp_hw_support/include/soc -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/esp_hw_support/port/esp32s3/. -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/esp_hw_support/port/esp32s3/private_include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/heap/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/log/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/lwip/include/apps -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/lwip/include/apps/sntp -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/lwip/lwip/src/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/lwip/port/esp32/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/soc/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/soc/esp32s3/. -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/soc/esp32s3/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/hal/esp32s3/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/hal/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/hal/platform_port/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/esp_rom/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/esp_rom/include/esp32s3 -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/esp_rom/esp32s3 -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/esp_common/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/esp_system/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/esp_system/port/soc -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/esp_system/port/public_compat -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/xtensa/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/xtensa/esp32s3/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/driver/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/driver/esp32s3/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/esp_pm/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/esp_ringbuf/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/efuse/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/efuse/esp32s3/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/vfs/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/esp_wifi/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/esp_event/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/esp_netif/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/esp_eth/include -I/home/me/Sync/bluepad32-esp-idf-arduino/esp/esp-idf/components/tcpip_adapter/include -I/home/me/Sync/bluepad32-esp [... cut ...]
/home/me/Sync/bluepad32-esp-idf-arduino/esp/my_project/main/sketch.cpp:276:5: error: 'Mouse' was not declared in this scope
     Mouse.begin();
     ^~~~~
/home/me/Sync/bluepad32-esp-idf-arduino/esp/my_project/main/sketch.cpp:276:5: note: suggested alternative: 'pause'
     Mouse.begin();
     ^~~~~
     pause
/home/me/Sync/bluepad32-esp-idf-arduino/esp/my_project/main/sketch.cpp:277:5: error: 'USB' was not declared in this scope
     USB.begin();
     ^~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

ricardoquesada, 2025-01-02:
What libraries are USB and USBHIDMouse ? Are they part of esp-idf ? or 3rd-party libs ?

If they are 3rd-party, you will need to add them in the `CMakeLists.txt`.

See:
https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/tree/develop?tab=readme-ov-file#using-3rd-party-arduino-libraries

ricardoquesada, 2025-01-03:
What you need to do if you use USBHID, etc...

https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html

controllercustom, 2025-01-03:
> What libraries are USB and USBHIDMouse ? Are they part of esp-idf ? or 3rd-party libs ?

USB.h and USBHIDMouse is included with the arduino-esp32 core.

```
./portable/packages/esp32/hardware/esp32/3.1.0/libraries/USB/src/USBHIDMouse.h
./portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/libraries/USB/src/USBHIDMouse.h
./portable/packages/esp32/hardware/esp32/3.1.0/cores/esp32/USB.h
./portable/packages/esp32-bluepad32/hardware/esp32/4.1.0/cores/esp32/USB.h
```

```
./esp/my_project/components/arduino/cores/esp32/USB.h
./esp/my_project/components/arduino/libraries/USB/src/USBHIDMouse.h
```

controllercustom, 2025-01-03:
OK, I will probably use Arduino IDE but patch the source file to remove the duplicate declaration. If a new version of the library is released I can reapply the patch.

ricardoquesada, 2025-01-03:
sure.
but if you want to try the template, read the link that I sent you