ricardoquesada/bluepad32 #132: Help with Bluetooth controller implementation

aedile, 2024-12-12:
Greetings,

I really like Galagino and I really like CYD but I find the Wiimote to be a little bit unwieldy for my purposes.  I thought I'd take a crack at getting the bluepad32 library working with galagino and I've mostly succeeded.  For now I've just made a custom implementation of Nunchuck.h.  Here is that file:

```
#ifndef _NUNCHUCK_H_
#define _NUNCHUCK_H_

#include <Bluepad32.h>

ControllerPtr gamepad = nullptr;

void onConnectedController(ControllerPtr gp) {
    gamepad = gp;
    Serial.println("Gamepad connected");
}

void onDisconnectedController(ControllerPtr gp) {
    if (gamepad == gp) {
        gamepad = nullptr;
        Serial.println("Gamepad disconnected");
    }
}

void nunchuckSetup() {
    delay(3000);
    BP32.setup(&onConnectedController, &onDisconnectedController);
    Serial.println("Bluepad32 setup completed");
}

unsigned char getNunchuckInput() {
  bool dataUpdated = BP32.update();
  if(gamepad && dataUpdated){
    Serial.printf("idx=%d, dpad: 0x%02x, buttons: 0x%04x, axis L: %4d, %4d, A: %d, B: %d, X: %d, Y: %d\n",
        gamepad->index(),        // Controller Index
        gamepad->dpad(),         // D-pad
        gamepad->buttons(),      // bitmask of pressed buttons
        gamepad->axisX(),        // (-511 - 512) left X Axis
        gamepad->axisY(),        // (-511 - 512) left Y axis
        gamepad->a(),            // A button
        gamepad->b(),            // B button
        gamepad->x(),            // X button
        gamepad->y()             // Y button
    );
    int joyY = gamepad->axisY();
    int joyX = gamepad->axisX();

    return ((joyX < -127) ? BUTTON_LEFT : 0) | //Move Left
            ((joyX >  127) ? BUTTON_RIGHT : 0) | //Move Right
            ((joyY < -127) ? BUTTON_UP : 0) | //Move Up
            ((joyY >  127) ? BUTTON_DOWN : 0) | //Move Down
            (gamepad->a() ? BUTTON_FIRE : 0) |
            (gamepad->b() ? BUTTON_EXTRA : 0) ;
  }else{
    Serial.println("No gamepad");
    return 0;
  }
  vTaskDelay(1);
}

#endif //_NUNCHUCK_H_
```

The cool thing is this is MOSTLY working.  When the library is compiled with the Bluepad32 version of ESP32 Dev Module, everything compiles, the controller is detected.  The buttons work and Up and Down work in the menu for picking which game you want to play.  The problem I'm having right now, and I'm just completely stuck on, is that, using the serial prints, I know I'm getting left and right from the x and y axes (-512 for left/up, 512 for rt/dn on press with it centered at maybe 3 on x axis and -3 on y axis).  I've even tried some debugging to verify that when dpad is pushed, that `getNunchuckInput` is returning a non-zero value different from normal with distinct values for left, right, up and down.  It seems to work fine in the menu, just not in any of the games.  Not sure what I am doing wrong here, unfortunately.  I feel like it's something to do with the return object.  Here is what the original Nunchuck.h returns:

```
return ((joyX < 127 - NUNCHUCK_MOVE_THRESHOLD) ? BUTTON_LEFT : 0) | //Move Left
           ((joyX > 127 + NUNCHUCK_MOVE_THRESHOLD) ? BUTTON_RIGHT : 0) | //Move Right
           ((joyY > 127 + NUNCHUCK_MOVE_THRESHOLD) ? BUTTON_UP : 0) | //Move Up
           ((joyY < 127 - NUNCHUCK_MOVE_THRESHOLD) ? BUTTON_DOWN : 0) | //Move Down
           (nchuk.buttonZ() ? BUTTON_FIRE : 0) |
           (nchuk.buttonC() ? BUTTON_EXTRA : 0) ;
  }
```
I feel like what I'm returning is not markedly different (just different expected values from the joystick).

Any hints?  Happy to clean up code and share here as a PR if I can get it working.  

Thanks!

aedile, 2024-12-12:
Example serial output dead stick:
```
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    3,   -3, A: 0, B: 0, X: 0, Y: 0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    3,   -3, A: 0, B: 0, X: 0, Y: 0

```

Example serial output left stick:
```
idx=0, dpad: 0x00, buttons: 0x0000, axis L: -512,   -3, A: 0, B: 0, X: 0, Y: 0
idx=0, dpad: 0x00, buttons: 0x0000, axis L: -512,   -3, A: 0, B: 0, X: 0, Y: 0
```

aedile, 2024-12-15:
Please close this if you get to it.  I meant to post it to Galagino issues board.  Thanks for Bluepad32.  It works great!!  I've used it on several projects now.  Cheers!

ricardoquesada, 2024-12-18:
ok. closing it.

closed 2024-12-18 by ricardoquesada