ricardoquesada/bluepad32 #133: [Bug]: Not compatible with other libraries

dwilliss, 2024-12-15:
### What happened?

Fails to compile when trying to incorporate other libraries into the sketch.
I have a library that I'm using on other ESP32 units to communicate between modules using ESP-NOW.

This tries to initialize WiFi, but the wifi stack included in Bluepad seems to be incompatible with the ESP32 3.0.7 version.  There are things missing.

### Bluepad32 Version

latest from develop branch

### Bluepad32 version custom

4.1.0

### Bluepad32 Platform

Arduino IDE

### Platform version

Arduino IDE 2.3.4

### Controller

PS4 DualShock clone

### Microcontroller

ESP32

### Microcontroller board

ESP32-Dev Kit

### OS

Windows

### Relevant log output

```shell
c:\Users\dwill\Documents\Arduino\libraries\EspNowChildNode\EspNowChildNode.cpp: In member function 'bool EspNowChildNode::begin()':
c:\Users\dwill\Documents\Arduino\libraries\EspNowChildNode\EspNowChildNode.cpp:59:12: error: 'class WiFiClass' has no member named 'setChannel'; did you mean 'channel'?
       WiFi.setChannel(ESPNOW_WIFI_CHANNEL, WIFI_SECOND_CHAN_NONE);
            ^~~~~~~~~~
            channel
c:\Users\dwill\Documents\Arduino\libraries\EspNowChildNode\EspNowChildNode.cpp:61:21: error: 'class WiFiClass' has no member named 'STA'
       while (!(WiFi.STA.started() || WiFi.AP.started())) {
                     ^~~
c:\Users\dwill\Documents\Arduino\libraries\EspNowChildNode\EspNowChildNode.cpp:61:43: error: 'class WiFiClass' has no member named 'AP'
       while (!(WiFi.STA.started() || WiFi.AP.started())) {
                                           ^~
c:\Users\dwill\Documents\Arduino\libraries\EspNowChildNode\EspNowChildNode.cpp:84:12: error: 'class WiFiClass' has no member named 'setChannel'; did you mean 'channel'?
       WiFi.setChannel(ESPNOW_WIFI_CHANNEL, WIFI_SECOND_CHAN_NONE);
            ^~~~~~~~~~
            channel
c:\Users\dwill\Documents\Arduino\libraries\EspNowChildNode\EspNowChildNode.cpp:86:21: error: 'class WiFiClass' has no member named 'STA'
       while (!(WiFi.STA.started() || WiFi.AP.started())) {
                     ^~~
c:\Users\dwill\Documents\Arduino\libraries\EspNowChildNode\EspNowChildNode.cpp:86:43: error: 'class WiFiClass' has no member named 'AP'
       while (!(WiFi.STA.started() || WiFi.AP.started())) {
```

### Relevant sketch

```shell
// SAVE THIS TO EspNowChildNode.h
#ifndef ESP_NOW_CHILD_NODE_H
#define ESP_NOW_CHILD_NODE_H
[... 338 lines cut ...]
// Example main sketch boiled down to just what wants to use this library

#include <EEPROM.h>

#include <EspNowChildNode.h>

void OnCommandReceived(const char* command) {
  Serial.println(command);
}

EspNowChildNode node(NODE_SOUND_BOARD);

char _inputBuffer[100];
int _inputOffset;

void setup() {
  _inputOffset = 0;
  Serial.begin(115200);
  if (!EEPROM.begin(6)) Serial.println("Can't init EEPROM?");
  EEPROM.begin(60);

  node.SetCallback(OnCommandReceived);
  node.begin();
}
```

ricardoquesada, 2025-01-01:
mmm... interesting.

If you try `develop` branch from https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template, does it work ?

It uses Arduino Core v3.1.0, and requires esp-idf 5.3.2.

gomer-noah, 2025-01-06:
I encountered the same problem trying to use ESP-NOW in conjunction with Bluepad32. Below is a much simpler sketch that also gives an error ( 'class WiFiClass' has no member named 'STA') in case that makes it easier to reproduce.

**Versions:**
Arduino IDE 2.3.4
ESP32 3.1.0
Bluepad32 4.1.0

**Board:**
ESP32 Dev Module

```
#include <WiFi.h>
#include <esp_wifi.h>

void readMacAddress(){
  uint8_t baseMac[6];
  esp_err_t ret = esp_wifi_get_mac(WIFI_IF_STA, baseMac);
  if (ret == ESP_OK) {
    Serial.printf("%02x:%02x:%02x:%02x:%02x:%02x\n",
                  baseMac[0], baseMac[1], baseMac[2],
                  baseMac[3], baseMac[4], baseMac[5]);
  } else {
    Serial.println("Failed to read MAC address");
  }
}

void setup(){
  Serial.begin(115200);

  WiFi.mode(WIFI_STA);
  WiFi.STA.begin();

  Serial.print("[DEFAULT] ESP32 Board MAC Address: ");
  readMacAddress();
}
 
void loop(){

}
```

I just needed to grab the MAC address so my workaround was to use the regular ESP32 libraries temporarily. It seems most basic ESP-NOW functionality still works with Bluepad32 so I didn't have a need to go the ESP-IDF route.

Blnko, 2025-03-25:
I uploaded the following sketch to esp32 Dev Kit v1 But my controller PS4 will not pair what step am I missing. Thank You
#include <Bluepad32.h>

ControllerPtr myControllers[BP32_MAX_GAMEPADS];

// This callback gets called any time a new gamepad is connected.
// Up to 4 gamepads can be connected at the same time.
void onConnectedController(ControllerPtr ctl) {
    bool foundEmptySlot = false;
    for (int i = 0; i < BP32_MAX_GAMEPADS; i++) {
        if (myControllers[i] == nullptr) {
            Serial.printf("CALLBACK: Controller is connected, index=%d\n", i);
            // Additionally, you can get certain gamepad properties like:
            // Model, VID, PID, BTAddr, flags, etc.
            ControllerProperties properties = ctl->getProperties();
            Serial.printf("Controller model: %s, VID=0x%04x, PID=0x%04x\n", ctl->getModelName().c_str(), properties.vendor_id,
                           properties.product_id);
            myControllers[i] = ctl;
            foundEmptySlot = true;
            break;
        }
    }
    if (!foundEmptySlot) {
        Serial.println("CALLBACK: Controller connected, but could not found empty slot");
    }
}

void onDisconnectedController(ControllerPtr ctl) {
    bool foundController = false;

    for (int i = 0; i < BP32_MAX_GAMEPADS; i++) {
        if (myControllers[i] == ctl) {
            Serial.printf("CALLBACK: Controller disconnected from index=%d\n", i);
            myControllers[i] = nullptr;
            foundController = true;
            break;
        }
    }

    if (!foundController) {
        Serial.println("CALLBACK: Controller disconnected, but not found in myControllers");
    }
}

void dumpGamepad(ControllerPtr ctl) {
    Serial.printf(
        "idx=%d, dpad: 0x%02x, buttons: 0x%04x, axis L: %4d, %4d, axis R: %4d, %4d, brake: %4d, throttle: %4d, "
        "misc: 0x%02x, gyro x:%6d y:%6d z:%6d, accel x:%6d y:%6d z:%6d\n",
        ctl->index(),        // Controller Index
        ctl->dpad(),         // D-pad
        ctl->buttons(),      // bitmask of pressed buttons
        ctl->axisX(),        // (-511 - 512) left X Axis
        ctl->axisY(),        // (-511 - 512) left Y axis
        ctl->axisRX(),       // (-511 - 512) right X axis
        ctl->axisRY(),       // (-511 - 512) right Y axis
        ctl->brake(),        // (0 - 1023): brake button
        ctl->throttle(),     // (0 - 1023): throttle (AKA gas) button
        ctl->miscButtons(),  // bitmask of pressed "misc" buttons
        ctl->gyroX(),        // Gyro X
        ctl->gyroY(),        // Gyro Y
        ctl->gyroZ(),        // Gyro Z
        ctl->accelX(),       // Accelerometer X
        ctl->accelY(),       // Accelerometer Y
        ctl->accelZ()        // Accelerometer Z
    );
}

void dumpMouse(ControllerPtr ctl) {
    Serial.printf("idx=%d, buttons: 0x%04x, scrollWheel=0x%04x, delta X: %4d, delta Y: %4d\n",
                   ctl->index(),        // Controller Index
[... 194 lines cut ...]
    // When enabled, controllers like DualSense and DualShock4 generate two connected devices:
    // - First one: the gamepad
    // - Second one, which is a "virtual device", is a mouse.
    // By default, it is disabled.
    BP32.enableVirtualDevice(false);
}

// Arduino loop function. Runs in CPU 1.
void loop() {
    // This call fetches all the controllers' data.
    // Call this function in your main loop.
    bool dataUpdated = BP32.update();
    if (dataUpdated)
        processControllers();

    // The main loop must have some kind of "yield to lower priority task" event.
    // Otherwise, the watchdog will get triggered.
    // If your main loop doesn't have one, just add a simple `vTaskDelay(1)`.
    // Detailed info here:
    // https://stackoverflow.com/questions/66278271/task-watchdog-got-triggered-the-tasks-did-not-reset-the-watchdog-in-time

    //     vTaskDelay(1);
    delay(150);
}
``

lorenzoconforti, 2025-05-27:
> hopefully it will get fixed once I released the new version for Arduino IDE.
> 
> In the meantime, you can try the "template" project which is already based on ESP-IDF v5.3 and Arduino Core 3.1
> 
> https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template

@ricardoquesada , any update on this?

Thank you