ricardoquesada/bluepad32 #134: feat: access raw input report and send raw output report structures

dragoncoder047, 2025-01-02:
continuing from https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/issues/11#issuecomment-2567217361 I figured I would get this in her before you release 4.2.0 ...

currently the functions available and exported (via .h files) are limited in that they can only send the functionality that has been defined already and known to work. however for someone who wants to experiment, not being able to send a raw bit array is quite frustrating and having to duplicate half of the code to be able get it to compile and send directly to a `uni_hid_device_t*` seems like a waste.

Can (for all controller types, but I'm mostly looking at PS5) the "xxx_send_output_report" functions and the "xxx_output_report_t" structure (+ all subtypes necessary to interpret it) be exported as well as the current functions? This would be nice, for example if there's something that is unimplemented right now, and I want to test it via an Arduino sketch before I fork BP32 and send a pull request.

Also, it would probably be harder to implement saving the most recent input report, but having access to  that would also help in decoding the unimplemented parts in the same manner as above.

https://github.com/ricardoquesada/bluepad32/blob/2152d0b7aba0b7cf5f54146e56b21c35dbb87024/src/components/bluepad32/parser/uni_hid_parser_ds5.c

https://github.com/ricardoquesada/bluepad32/blob/2152d0b7aba0b7cf5f54146e56b21c35dbb87024/src/components/bluepad32/include/parser/uni_hid_parser_ds5.h

ricardoquesada, 2025-01-02:
I might be missing something. You just need to call `uni_hid_device_send_ctrl_report(uni_hid_device_t* d, const uint8_t* report, uint16_t len);` which is public

dragoncoder047, 2025-01-02:
That would work but it means you have to copy the code that is already implemented, to construct the report. at least if `xxx_output_report_t` was public you could create a new one, assign the fields you need, cast it to `uint8_t*` and send it.

ricardoquesada, 2025-01-02:
yep. Making the report struct public means that I have to maintain it forever for backward compatibility.

If in the future I want to add or remove something, it might break code that uses it.

I suggest that you copy the report struct, and/or create your own.

dragoncoder047, 2025-01-02:
ok, I guess I can just copy those data stuff and send it.

but is there a way to get the "state" of  the gamepad in something future proof? Because of stuff like this:

https://github.com/ricardoquesada/bluepad32/blob/2152d0b7aba0b7cf5f54146e56b21c35dbb87024/src/components/bluepad32/parser/uni_hid_parser_ds5.c#L744-L745

if I choose the wrong sequence id, the report will silently fail, or if I happen to choose the right one then BP32 won't get updated and use the same sequence number on the next report and it will get messed up, etc. etc.

ricardoquesada, 2025-01-02:
TL;DR: short term solution: copy & paste the function from `uni_hid_parser_ds5.c`

Yeah, in your case I suggest that you use the "template" project.

The template project comes with the bluepad32 source code, as opposed to the "Bluepad32 Arduino board" that comes with bluepad32 pre-compiled.

since you are experimenting with the internals of bluepad32, you need to have access to the source code and change it as you need.

I'm not opposed to making some functions public, but I need to have the "full picture" of what it is needed. And not a case-by-case basic.

For context: The "template" project uses the ESP32 Arduino  Core... so you can use all your Arduino Core.

What I might do in the future, is to release the "Bluepad32 Arduino board" with the Bluepad32 source code instead of the pre-compiled version.

dragoncoder047, 2025-01-02:
OK, maybe I'll give PlatformIO a try again. I had shied away from it in the past because I could not get it to install the ESP32 toolchain in a way that worked (I kept getting "compiler not found" errors even though it was installed and on my PATH too), but it's been a few years since that happened so maybe I'll give it another try.

ricardoquesada, 2025-01-02:
checkout `develop` branch of the "template" project, and follow the `README.md` instructions for PlatformIO.

You might need to remove ALL your PlatformIO cache (e.g: `rm -rf ~/.platfromio`).

If it doesn't work, LMK.

I tested it on Linux and macOS... wouldn't know whether it works on Windows