ricardoquesada/bluepad32 #135: [Bug]: cannot connect more than 1 controller

davidos81, 2025-01-05:
### What happened?

I have a few different bluetooth controllers , a wii controller , ps4 copy controller and a game sir nova lite , all of them work 100% fine when paired , however if I try to connect more than 1 then the secound one will not pair

So I thought maybe they don't connect because of different manufacturers so I bought a 2nd gamesir nova lite , and that too works but only on its own

How do i get the example arduino project to work with more than 1 controller ?

thanks in advance , and well done on such a brilliant project!

### Bluepad32 Version

Other

### Bluepad32 version custom

bluepad32 4.1.0

### Bluepad32 Platform

Arduino IDE

### Platform version

arduino 2.34

### Controller

issues

### Microcontroller

ESP32

### Microcontroller board

esp32 dev module

### OS

Windows

### Relevant sketch

```shell
#include <Bluepad32.h>

ControllerPtr myControllers[BP32_MAX_GAMEPADS];

// This callback gets called any time a new gamepad is connected.
// Up to 4 gamepads can be connected at the same time.
void onConnectedController(ControllerPtr ctl) {
    bool foundEmptySlot = false;
    for (int i = 0; i < BP32_MAX_GAMEPADS; i++) {
        if (myControllers[i] == nullptr) {
            Serial.printf("CALLBACK: Controller is connected, index=%d\n", i);
            // Additionally, you can get certain gamepad properties like:
            // Model, VID, PID, BTAddr, flags, etc.
            ControllerProperties properties = ctl->getProperties();
            Serial.printf("Controller model: %s, VID=0x%04x, PID=0x%04x\n", ctl->getModelName().c_str(), properties.vendor_id,
                           properties.product_id);
            myControllers[i] = ctl;
            foundEmptySlot = true;
            break;
        }
    }
    if (!foundEmptySlot) {
        Serial.println("CALLBACK: Controller connected, but could not found empty slot");
    }
}
[... 238 lines cut ...]
    // When enabled, controllers like DualSense and DualShock4 generate two connected devices:
    // - First one: the gamepad
    // - Second one, which is a "virtual device", is a mouse.
    // By default, it is disabled.
    BP32.enableVirtualDevice(false);
}

// Arduino loop function. Runs in CPU 1.
void loop() {
    // This call fetches all the controllers' data.
    // Call this function in your main loop.
    bool dataUpdated = BP32.update();
    if (dataUpdated)
        processControllers();

    // The main loop must have some kind of "yield to lower priority task" event.
    // Otherwise, the watchdog will get triggered.
    // If your main loop doesn't have one, just add a simple `vTaskDelay(1)`.
    // Detailed info here:
    // https://stackoverflow.com/questions/66278271/task-watchdog-got-triggered-the-tasks-did-not-reset-the-watchdog-in-time

        //  vTaskDelay(1);
    delay(150);
}
```

davidos81, 2025-01-07:
forgot to add console output
I get the following line every secound as normal when 1 controller is connected
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,    0, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, etc..

but then every few secounds i get this:

```
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,    0, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y: -4095 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,    0, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y: -4095 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,    0, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y: -4095 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,    0, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y: -4095 z:     0
Failed to inquiry name for 6F:8B:AB:8B:60:9C, using a fake one
uni_hid_device_disconnect: Unknown GAP connection type: 0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,    0, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y: -4095 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,    0, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y: -4095 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,    0, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     ``
```

ricardoquesada, 2025-01-08:
weird. which board are you using ?
Do you have spare one to try on it ?

sometimes it could be a bug in esp32

davidos81, 2025-01-09:
@ricardoquesada thankyou for getting your kind reply!

I have no idea whats going on , so like you said use a different board - So i used my ODROID GO ESP32 edition and that uses a WROVER esp32 - BUT - still same thing , the thing is it works flawlessly with 1 controller and I really want to use it in my up and coming commercial product but I need that additional controller support , 

Do i need a new(ish) classic esp32? or maybe arduino settings ?

thanks in advance

ricardoquesada, 2025-01-09:
to discard possible bugs (and to have access to the console) I suggest that you try the esp32 example based on esp-idf (not arduino).
DO NO MODIFY the example. Just run it. and try again.
And if it stills fails, copy & paste the output from the console.

Example & instructions here:
https://github.com/ricardoquesada/bluepad32/tree/main/examples/esp32

davidos81, 2025-01-09:
ok and thanks for that!
I will give it a try tonight and let you know !

davidos81, 2025-01-09:
ok , so i tried the IDF example as you suggested and can confirm it did work correctly with 2 controllers !! , I still need to do stability & performance etc , but this is fantastic that it works  

So that begs the question , why did this not work with ARDUINO? 

But for now thanks @ricardoquesada

davidos81, 2025-01-10:
@ricardoquesada actually that was the one I got working , am not an IDF person myself , seldom use it , but for now it works exactly as I expected (my_project example), though does mean i have to rewrite code for IDF if I want this to work inside my project - but thats another thing.

In your opinion which is the least powerful chip in the esp32 series for classic bluetooth , as am thinking of simply having bluepad32 on its own mcu and it sends commands to my esp32 s3/s4 through UART, so a single core variant will do the task and be low cost

again , superb work @ricardoquesada  ,

ricardoquesada, 2025-01-10:
I wouldn't use two microcontrollers... that will add an extra layer of complexity.

In any case, if "esp-idf" is too complex, try PlatformIO... which simplifies things a bit.

But if you really want to use two microcontrollers, get the cheapeast ESP32 that you can find. All have similar specs... what changes is the number of GPIOs and/or RAM (???).

You might want to take a look at the AirLift module and how Bluepad32 interacts with it: https://bluepad32.readthedocs.io/en/latest/plat_airlift/

davidos81, 2025-01-10:
many thanks again @ricardoquesada !

 am on windows and can never for the life of me get platformio and IDF to work together , so I just use platformio with arduino libraries , and arduino for testing new sketches.

So I will struggle to get my stuff converted to IDF , but its not a big deal , the pros of getting multiple reliable bluetooth controllers vs a smallish layer of complexity (using a seperate mcu) is well worth it

I am glad i came across this brilliant library you have made!

I think i will close this , as actually the issue is solved at least on IDF, am sure arduino will be fixed soon !  

many thanks @ricardoquesada !!

closed 2025-01-10 by davidos81