ricardoquesada/bluepad32 #137: [Bug]: ASSERT failure in rwbt.c

vortex314, 2025-01-11:
### What happened?

Bluepad32 on espidf is working fine for several minutes , no noticeable memory leak. The connection and reception of data with a PS4 controller works fine.
Then it crashes 
```
I 00:08:28.913 |        main.cpp: 184 |  free heap size: 34144
I 00:08:29.921 |        main.cpp: 184 |  free heap size: 34396
I 00:08:30.945 |        main.cpp: 184 |  free heap size: 34380
I 00:08:31.953 |        main.cpp: 184 |  free heap size: 35104
ASSERT_PARAM(8192 0), in rwbt.c at line 360
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4008ebfc: f01d020c 00004136 f01d0000
Core  0 register dump:
PC      : 0x4008ec03  PS      : 0x00060034  A0      : 0x800884da  A1      : 0x3ffc2108  
A2      : 0x00000000  A3      : 0x00002000  A4      : 0x00000000  A5      : 0x3ff9e7d2  
A6      : 0x00000168  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc2078  
A10     : 0x00000000  A11     : 0x3ffc209b  A12     : 0x3ffc2047  A13     : 0x00000030  
A14     : 0x00000000  A15     : 0x3ffc204c  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4008ec00:0x3ffc2108 |<-CORRUPTED

ELF file SHA256: 77db749cd

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15564
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3904
entry 0x40080638
I (31) boot: ESP-IDF 5.3.0 2nd stage bootloader
I (31) boot: compile time Jan 10 2025 22:41:09
```

### Bluepad32 Version

I don't know

### Bluepad32 version custom

The github main branch.

### Bluepad32 Platform

ESP-IDF

### Platform version

ESP-IDF 5.3.0

### Controller

A PS4 clone controller

### Microcontroller

ESP32

### Microcontroller board

ESP32 Devkit V1
Purpose is to catch PS4 events / commands and send them via Wifi to the Zenoh middleware.

### OS

None

### Relevant log output

```shell
Platformio.ini file

[env:az-delivery-devkit-v4]
platform = espressif32
board = az-delivery-devkit-v4
framework = espidf
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
lib_deps = 
	https://github.com/eclipse-zenoh/zenoh-pico
	https://github.com/bergzand/NanoCBOR
	bblanchon/ArduinoJson@^7.3.0
build_flags = 
	-DWIFI_SSID=${sysenv.WIFI_SSID}
	-DWIFI_PASS=${sysenv.WIFI_PASS}
	-DZENOH_ESPIDF
	-DZ_FEATURE_LINK_UDP_MULTICAST=1
	-DZENOH_DEBUG=1
	-DBATCH_UNICAST_SIZE=1024
	-DBATCH_MULTICAST_SIZE=1024
	-DZ_FEATURE_AUTO_RECONNECT=1
	-DFRAG_MAX_SIZE=1024
	-Wmissing-field-initializers
	-std=gnu++17
	-Wno-error=deprecated-declarations
	-DCONFIG_TCPIP_LWIP=1
	-L/home/lieven/workspace/bluepad32/examples/esp32/build/esp-idf/btstack
	-lbtstack
	-Os
build_unflags = 
	-std=gnu++11

board_upload.flash_size = 2MB
board_upload.maximum_size = 2048000
board_build.partitions = huge_app.csv
```

### Relevant sketch

```shell
https://github.com/vortex314/zenoh-projects/tree/main/zenoh-esp32

The main PS4 code is in the ps4_actor.cpp
```

vortex314, 2025-01-11:
I'm guessing the cause is somewhere a stack overflow.

vortex314, 2025-01-11:
Hi Ricardo, thanks for the swift reaction. 
The reproduction of this issue will not be simple, the segfault occurs in some code rwbt.c, which is according to my own research part of the espressif library. 
The issue can reproduce after a few minutes or even after 44 min, my current guess is that it is coming from the combination of Wifi and BT stack.  So the sample Bluepad32 code will not help. My own notes -- https://github.com/vortex314/zenoh-projects/blob/main/zenoh-esp32/issue_rwbt.md
Will update here my findings. 
To be continued...

ricardoquesada, 2025-01-11:
try to get the stacktrace if possible.
Here the   info how to get it:

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/core_dump.html

vortex314, 2025-01-12:
````
lieven@pcpro:~/workspace/zenoh-projects/zenoh-esp32$ idf.py coredump-info
/home/lieven/esp/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: coredump-info
Serial port /dev/ttyUSB1
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Path to core dump file is not provided. Core dump can't be read from flash since this option is not enabled in menuconfig
lieven@pcpro:~/workspace/zenoh-projects/zenoh-esp32$ 
```
Suggestions ? It core deumped after 50 min runnning.

vortex314, 2025-01-12:
AT reboot
```
I (799) esp_core_dump_flash: Init core dump to flash
I (805) esp_core_dump_flash: Found partition 'coredump' @ 180000 131072 bytes
I (851) esp_core_dump_flash: Core dump data checksum is correct
I (851) esp_core_dump_flash: Found core dump 28132 bytes in flash @ 0x180000
I (855) coexist: coex firmware version: 4482466
I (861) main_task: Started on CPU0
I (864) main_task: Calling app_main()
```

vortex314, 2025-01-12:
Ok, it seems that my use of platformio makes also use of another sdkconfig file, when copying this pio file to the sdkconfig , I got a step further 
```
lieven@pcpro:~/workspace/zenoh-projects/zenoh-esp32$ idf.py coredump-info
/home/lieven/esp/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: coredump-info
Serial port /dev/ttyUSB1
Connecting.........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
WARNING:root:The core dump image offset is not specified. Use partition offset: 0x180000.
Failed to load core dump: Invalid application image for coredump: coredump SHA256(17d4c9f2a) != app SHA256().
lieven@pcpro:~/workspace/zenoh-projects/zenoh-esp32$ 
```

vortex314, 2025-01-12:
Finally, found something that works. I guess the tools don´t take into account the platformio locations 
```
lieven@pcpro:~/workspace/zenoh-projects/zenoh-esp32$ espcoredump.py info_corefile  -t raw  .pio/build/az-delivery-devkit-v4/firmware.elf
INFO: Invoke esptool to read image.
INFO: Retrieving core dump partition offset and size...
INFO: Core dump partition offset=1572864, size=131072
WARNING: The core dump image offset is not specified. Use partition offset: 0x180000.
INFO: esptool.py v4.7.0
Found 33 serial ports
Serial port /dev/ttyUSB1
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision v0.0)
Features: WiFi, BT, Dual Core, Coding Scheme None
Crystal is 40MHz
MAC: 30:ae:a4:03:ec:98
Uploading stub...
Running stub...
Stub running...
16 (100 %)
16 (100 %)
Read 16 bytes at 0x00180000 in 0.0 seconds (12.5 kbit/s)...
Hard resetting via RTS pin...

INFO: esptool.py v4.7.0
Found 33 serial ports
Serial port /dev/ttyUSB1
Connecting......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting..........
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision v0.0)
Features: WiFi, BT, Dual Core, Coding Scheme None
Crystal is 40MHz
MAC: 30:ae:a4:03:ec:98
Uploading stub...
Running stub...
Stub running...
28132 (100 %)
28132 (100 %)
Read 28132 bytes at 0x00180000 in 2.5 seconds (88.9 kbit/s)...
Hard resetting via RTS pin...

===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffe2a4c, name: 'wifi', GDB name: 'process 1073621580'
Crashed task is not in the interrupt context

================== CURRENT THREAD REGISTERS ===================
exccause       0xffff (InvalidCauseRegister)
excvaddr       0x0
pc             0x20000000          0x20000000
lbeg           0x4000c2e0          1073791712
lend           0x4000c2f6          1073791734
lcount         0xffffffff          4294967295
sar            0x17                23
ps             0x30                48
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
[... 275 lines cut ...]
.coredump.tasks.data 0x3ffdd7d0 0x210 RW 
.coredump.tasks.data 0x3fff8548 0x154 RW 
.coredump.tasks.data 0x3fff8380 0x1c0 RW 
.coredump.tasks.data 0x3ffe20a0 0x154 RW 
.coredump.tasks.data 0x3ffe1e30 0x260 RW 
.coredump.tasks.data 0x3ffe3ba8 0x154 RW 
.coredump.tasks.data 0x3ffe3990 0x210 RW 
.coredump.tasks.data 0x3ffe9294 0x154 RW 
.coredump.tasks.data 0x3ffe90a0 0x1e0 RW 
.coredump.tasks.data 0x3ffb7318 0x154 RW 
.coredump.tasks.data 0x3ffb7160 0x1b0 RW 
.coredump.tasks.data 0x3ffdec88 0x154 RW 
.coredump.tasks.data 0x3ffdea60 0x220 RW 
.coredump.tasks.data 0x3ffb7a50 0x154 RW 
.coredump.tasks.data 0x3ffb7890 0x1b0 RW 
.coredump.tasks.data 0x3fff6fec 0x154 RW 
.coredump.tasks.data 0x3fff6b70 0x470 RW 
.coredump.tasks.data 0x3ffee618 0x154 RW 
.coredump.tasks.data 0x3ffee410 0x200 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
lieven@pcpro:~/workspace/zenoh-projects/zenoh-esp32$ 
```

vortex314, 2025-01-12:
I got additional coredumps, but it seems that the task that crashes hasn 't any stack trace.  In all different cases

ricardoquesada, 2025-01-13:
so `rwbt.c` (If I have to guess it might be related with "Read Write Bluetooth") seems to be a blob.. no source code for that, and I wouldn't know why it is crashing.

yeah, better to keep the discussion on the Espressif bug.

VERY IMPORTANT:  ONLY call Bluepad32 / btstack from thread "thread 11" (ps4). NEVER call it from another thread.

If you need to use this tip:
https://bluepad32.readthedocs.io/en/latest/programmers_guide_raw/#multithreading

vortex314, 2025-01-13:
Thanks for your assistance, I  close the issue here.
 I suspect the explanation needs to come from the code of espressif.
I have not much knowledge on BT connectivity, but it looks to consume a lot of memory with high peaks. Squeezing the code of Wifi and BT together in IRAM is a challenge already.

vortex314, 2025-01-13:
@ricardoquesada  seen the feedback from Espressif on the version, I start to suspect a version issue. 

The bluepad32 doc : https://bluepad32.readthedocs.io/en/latest/plat_esp32/ specifies to execute a script from 'btstack' which replaces the btstack of espressif with another one . https://github.com/bluekitchen/btstack/blob/5d4d8cc7b1d35a90bbd6d5ffd2d3050b2bfc861c/port/esp32/integrate_btstack.py   -- 3 years old.
 Is this still necessary ? And also to apply patches.

ricardoquesada, 2025-01-13:
yes, but do it the "bp32" way. Explained here:

https://bluepad32.readthedocs.io/en/latest/plat_esp32/#3-patch-btstack-and-integrate-it-as-a-local-component

ak1932, 2025-04-05:
@vortex314  did you manage to fix this? I am encountering a very similar issue. @ricardoquesada, could the problem be because we are using different esp idf versions than the bluepad32 was designed for? If so, what version is suggested to be used? Any help would be great.

vortex314, 2025-04-05:
@ak1932  see the details of the espressif issue https://github.com/espressif/esp-idf/issues/15190 , I got it stabilized with changing some parameters, but if I dare to add some code, it became unstable again.
I think there is a fundamental issue with the espressif rom code , when the same radio is used for Wifi and Bluetooth, it needs to switch continuously. So I am guessing there is some race condition that ,even after hours of correct functioning ,  crashes the system.

ak1932, 2025-04-05:
> [@ak1932](https://github.com/ak1932) Perhaps we can try again with esp-idf 5.3 or 5.4

I am currently on v5.4. I have a few questions, what idf does the arduino core for bluepad32 use? As I used the library on arduino a while back and didn't encounter any such crashes. Was this purely luck and I would have crashed on arduino as well, or is it worth trying to downgrade my idf version.

My arduino program was relatively simple and current idf firmware is a lot more complicated, so it would be a huge hassle to try to convert it to arduino in the off chance it works better.

Anyways, thanks for the great work @ricardoquesada ! As I understand it is an issue in the esp firmware itself and there is only so much that can be done from outside.

ricardoquesada, 2025-04-06:
The BP32  "Arduino IDE" version is using ESP-IDF 4.4 (IIRC), which uses the old ESP-IDF For Arduino v2.3.

If you want to try with Arduino + ESP-IDF 5.3, use the "Template" version: https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template