ricardoquesada/bluepad32 #140: [Bug]: no way to access HID_USAGE_FAST_FORWARD, HID_USAGE_REWIND... which are sent by the keyboard

negrusti, 2025-01-24:
### What happened?

These are not available for the isKeyPressed(KeyboardKey key), and there is no generic function to get the pressed button code apparently?

### Bluepad32 Version

latest from develop branch

### Bluepad32 Platform

ESP-IDF

### Platform version

PlatformIO with ESP-IDF

### Controller

5-button keyboard from AliExpress. 3 buttons work and "forwards", "backwards" do not.

### Microcontroller

ESP32

### Microcontroller board

ESP32-S3-WROOM1 DevKit from Lolin. This is the store URL: https://example.com

### OS

Windows

### Relevant log output

```shell

```

### Relevant sketch

```shell

```

ricardoquesada, 2025-01-24:
Please, add link to the keyboard that is not working

negrusti, 2025-01-24:
@ricardoquesada https://vi.aliexpress.com/item/1005004899620476.html
all 5 keys trigger isKeyPressed(), but for + and - there is no value

idx=0, Pressed keys:                     for (-)
idx=0, Pressed keys:                     for (+)
idx=0, Pressed keys: LeftArrow,
idx=0, Pressed keys: RightArrow,
idx=0, Pressed keys: Pause,

negrusti, 2025-01-25:
These were VolumeUp, VolumeDown, apologies. Closing this.

closed 2025-01-25 by negrusti