ricardoquesada/bluepad32 #141: [Bug]: Cannot connect keyboard

magarto, 2025-01-29:
### What happened?

Hi,

Trying to connect a Logitech K380 it doesn't allow me to type the numeric value confirmation to pair it.
Is there any way to get it working? Even if I have to type de MAC and hex key manually.

> bp32> Device found: F4:73:35:A1:44:CC with COD: 0x002540, pageScan 1, clock offset 0x35b4, rssi 209 dBm
> Creating device: F4:73:35:A1:44:CC (idx=0)
> uni_bt_process_fsm, bd addr:F4:73:35:A1:44:CC,  state: 1, incoming:0
> uni_bt_process_fsm: requesting name
> --> HCI_EVENT_REMOTE_NAME_REQUEST_COMPLETE
> Name: 'Keyboard K380'
> uni_bt_process_fsm, bd addr:F4:73:35:A1:44:CC,  state: 4, incoming:0
> uni_bt_process_fsm: Starting L2CAP connection
> --> HCI_EVENT_CONNECTION_COMPLETE
> SM_EVENT_IDENTITY_RESOLVING_STARTED
> Identity resolving failed for F4:73:35:A1:44:CC
> 
> --> HCI_EVENT_LINK_KEY_REQUEST:
> SSP User Confirmation Request with numeric value '39038'
> SSP User Confirmation Auto accept
> --> HCI_EVENT_AUTHENTICATION_COMPLETE_EVENT: status=0, handle=0x0081
> L2CAP_EVENT_CHANNEL_OPENED (channel=0x0041)
> PSM: 0x0011, local CID=0x0041, remote CID=0x0043, handle=0x0081, incoming=0, local MTU=1691, remote MTU=48, addr=F4:73:35:A1:44:CC
> HID Control opened, cid 0x41
> uni_bt_process_fsm, bd addr:F4:73:35:A1:44:CC,  state: 10, incoming:0
> uni_bt_process_fsm: Create L2CAP interrupt connection
> L2CAP_EVENT_CHANNEL_OPENED (channel=0x0042)
> PSM: 0x0013, local CID=0x0042, remote CID=0x0044, handle=0x0081, incoming=0, local MTU=1691, remote MTU=48, addr=F4:73:35:A1:44:CC
> HID Interrupt opened, cid 0x42
> Device F4:73:35:A1:44:CC is connected
> uni_bt_process_fsm, bd addr:F4:73:35:A1:44:CC,  state: 12, incoming:0
> uni_bt_process_fsm: starting SDP query
> -----------> sdp_query_start()
> Starting SDP VID/PID query for F4:73:35:A1:44:CC
> Device found: F4:73:35:A1:44:CC with COD: 0x002540, pageScan 1, clock offset 0x35b4, rssi 209 dBm
> Device already added, waiting (current state=0x05)...
> uni_bt_process_fsm, bd addr:F4:73:35:A1:44:CC,  state: 5, incoming:0
> Vendor ID: 0x046d - Product ID: 0xb342
> Device (vendor_id=0x046d, product_id=0xb342) not found in DB.
> Device detected as Keyboard: 0x190
> uni_bt_process_fsm, bd addr:F4:73:35:A1:44:CC,  state: 6, incoming:0
> uni_bt_process_fsm: querying HID descriptor
> Starting SDP HID-descriptor query for F4:73:35:A1:44:CC
> <------- sdp_query_timeout()
> Failed to query SDP for F4:73:35:A1:44:CC, timeout
> Device cannot connect in time, deleting:
> 	btaddr: F4:73:35:A1:44:CC
> 	bt: handle=129 (ACL), hids_cid=0, ctrl_cid=0x0041, intr_cid=0x0042, cod=0x00002540, flags=0x00003b00, incoming=0
> 	model: vid=0x046d, pid=0xb342, model='Keyboard', name='Keyboard K380'
> 	battery: 0 / 255, type=unknown
> 	uni_hid_parser_keyboard_device_dump: implement me: 
> Disconnecting device: F4:73:35:A1:44:CC
> Deleting device: F4:73:35:A1:44:CC
> L2CAP_EVENT_CHANNEL_CLOSED: 0x0042 (channel=0x0042)
> Couldn't not find hid_device for cid = 0x0042
> L2CAP_EVENT_CHANNEL_CLOSED: 0x0041 (channel=0x0041)
> Couldn't not find hid_device for cid = 0x0041
> --> HCI_EVENT_DISCONNECTION_COMPLETE

Thanks

### Bluepad32 Version

3.10.3

### Bluepad32 version custom

Example: Using Git develop branch commit hash #xxxxxxx

### Bluepad32 Platform

None

### Platform version

ESP-IDF v4.3.2

### Controller

Xbox Wireless 1708 using firmware 5.15.1923. Link to controller: https://example.com

### Microcontroller

ESP32

### Microcontroller board

ESP32-S3-WROOM1

### OS

Linux

### Relevant log output

```shell

```

### Relevant sketch

```shell

```

ricardoquesada, 2025-01-29:
what bluepad32 version are you using ? 
Try to use v4.2 which has some BLE related bug fixes

magarto, 2025-01-29:
Yes, I'm using bluepad32 4.2.0 version

ricardoquesada, 2025-01-29:
It is failing because it cannot parse the SDP... mmm, so probably it is not a BLE device.

it seems that the key was accepted.

magarto, 2025-01-30:
Its a Logitech K380, I think It is not BLE but Bluetooth 3.0.
Is there any way to make it work? Maybe disabling BLE with menuconfig?

ricardoquesada, 2025-01-30:
If I could get one of those I might (big **might**) be able to make it work (see: https://bluepad32.readthedocs.io/en/latest/FAQ/#could-you-add-support-for-my-unsupported-controller )

In the meantime, the supported keyboard are here:
https://bluepad32.readthedocs.io/en/latest/supported_keyboards/

magarto, 2025-01-30:
Ok. Thanks and congrats for this project