ricardoquesada/bluepad32 #148: [Bug]: Game controller connects but no input received

Monnandre, 2025-03-14:
### What happened?

The esp32 successfully detects the gamepad (SFR Manette Plus) but fails to get inputs from it.
When pairing the gamepad with my phone and using GamePad Tester, it works and is recognized as an Android Gamepad (also works with PS4 Mode Type 2)

### Bluepad32 Version

Other

### Bluepad32 version custom

Bluepad32 version 4.1.0

### Bluepad32 Platform

Arduino IDE

### Platform version

Arduino IDE 2.3.2

### Controller

SFR Manette Plus https://www.sfr.fr/accessoires/details/la-manette-plus

### Microcontroller

ESP32

### Microcontroller board

ESP32-DevKit v1

### OS

Windows

### Relevant log output

```shell
21:11:21.626 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:11:21.626 -> mode:DIO, clock div:1
21:11:21.626 -> load:0x3fff0030,len:1184
21:11:21.626 -> load:0x40078000,len:13260
21:11:21.626 -> load:0x40080400,len:3028
21:11:21.626 -> entry 0x400805e4
21:11:21.942 -> Bluepad32 (C) 2016-2024 Ricardo Quesada and contributors.
21:11:21.984 -> Version: v4.1.0
21:11:21.984 -> BTstack: Copyright (C) 2017 BlueKitchen GmbH.
21:11:22.584 -> BTstack up and running at 5C:01:3B:89:A9:46
21:11:22.617 -> Firmware: Bluepad32 for Arduino v4.1.0
21:11:22.617 -> BD Addr: 5C: 1:3B:89:A9:46
21:11:22.617 -> C1 A6 BC 53 DD E2 A2 A8 04 37 0A 68 03 A6 AE 1E 
21:11:49.596 -> CALLBACK: Controller is connected, index=0
21:11:49.641 -> Controller model: XBox 360, VID=0x0000, PID=0x0000
21:11:49.641 -> idx=0, dpad: 0x00, buttons: 0x0000, axis L:    0,    0, axis R:    0,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
21:11:49.768 -> idx=0, dpad: 0x00, buttons: 0x0000, axis L:    0,    0, axis R:    0,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
21:11:49.936 -> idx=0, dpad: 0x00, buttons: 0x0000, axis L:    0,    0, axis R:    0,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
```

### Relevant sketch

```shell
exemple script controller.ino
```

sufiaji, 2025-04-03:
dear team, 
thank you for this great work! however I encounter the same issue. Here is my gamepad: 

https://id.aliexpress.com/item/1005005935291995.html?gatewayAdapt=glo2idn

Bluepad32 (C) 2016-2024 Ricardo Quesada and contributors.
Version: v4.1.0
BTstack: Copyright (C) 2017 BlueKitchen GmbH.
BTstack up and running at 98:F4:AB:6D:6E:86
Firmware: Bluepad32 for Arduino v4.1.0
BD Addr: 98:F4:AB:6D:6E:86
CALLBACK: controller is connected
Controller Model: Android, VID=0x1949, PID:0x0402

then all the values are 0 at anytime

Hopefully there will be an update soon 👍

ricardoquesada, 2025-04-09:
yep. some clones have some issues.

here is the list of supported gamepads: https://bluepad32.readthedocs.io/en/latest/supported_gamepads/

take a look at the "misc" section.

and if you want me to add support for any particular gamepad that is not on the list, please read:

https://bluepad32.readthedocs.io/en/latest/FAQ/#could-you-add-support-for-my-unsupported-controller