ricardoquesada/bluepad32 #150: [Bug]: accelerometer, Gyro and battery showing 0

Robotadho, 2025-03-23:
### What happened?

A bug happened! I'm connecting my PS4 con (Aftermarket) to Esp WROOM 32. all the button and joystick are working find. but gyro, accell , and battey shown 0. I check my controller with DS4 checker all are working find. please help. 

[image]

### Bluepad32 Version

latest from develop branch

### Bluepad32 Platform

Arduino IDE

### Platform version

Version: 2.3.4
Date: 2024-12-03T10:03:03.490Z
CLI Version: 1.1.1

### Controller

PS4 controller (aftermarket)

### Microcontroller

ESP32

### Microcontroller board

ESP32-S DOIT WIFI. This is the store URL: https://www.tokopedia.com/cncstorebandung/esp32-s-doit-wifi-bluetooth-iot-dual-core-type-c-38p-plus-kabel-data

### OS

Windows

### Relevant log output

```shell

```

### Relevant sketch

```shell
Bluepad32 example
```

ricardoquesada, 2025-03-24:
yep. probably related to not being a genuine DS4 controller.

closed 2025-03-24 by ricardoquesada

solhuebner, 2025-03-25:
@Robotadho do you have a link for the controller?