ricardoquesada/bluepad32 #152: Add Codecell library

da43-hub, 2025-03-27:
Hi!  I realy appreceate all you have created with bluepad32. I have a project of a microcar with a CodeCell eso 32devc3 i want ti control via my xbox controller using bluepad library and codecell library.if i chek your controller example under esp32+bluepad32 under esp32devc3 it works however i can not load the codecell.h library. If  i run thw oppisite under esp32 under esp32devc3 i can not load the bluepad library.

I contacted the codecell team and this was their response:"Hi David,i see so you are not selecting the esp32c3dev board from the esp32 board manager - the issue might be that the esp32+bluepad32 might not be updated with the latest drivers for the esp32 - is the error reported related to timers?

in that case we suggest getting in touch with the bluepad developers to see if they can update there drivers - from our end we keep our library updated using the latest esp32 drivers so we cannot downgrade"

https://github.com/microbotsio/CodeCell

So im humbally asking for your help here so my son can enjoy xbox control..

Thanks,
David

ricardoquesada, 2025-04-19:
Please read: http://bit.ly/bluepad32-help

- What "flavor" of bluepad32 are you using ? (Arduino IDE, PlatformIO, etc...)
- What version ?

the more details you add, the easier is going to be to help you.

da43-hub, 2025-04-28:
Hi,
Thank you for your reply!
i am using option A, which is  Arduino IDE
"Arduino Core for ESP32 + Bluepad32" board, version 4.1.0 installed.

When i even run your example(that works) and only add the codecell library
with no code i get an error message, see below:

error l;og trying to compile:
c:\Users\david\Documents\Arduino\libraries\CodeCell\src\CodeCell.cpp: In
member function 'void CodeCell::Init(uint16_t)':
c:\Users\david\Documents\Arduino\libraries\CodeCell\src\CodeCell.cpp:73:33:
error: 'RMT_MEM_NUM_BLOCKS_1' was not declared in this scope
   rmtInit(LED_PIN, RMT_TX_MODE, RMT_MEM_NUM_BLOCKS_1, 10000000);
/*Configure RMT to run the onboard addressable LED*/
                                 ^~~~~~~~~~~~~~~~~~~~
c:\Users\david\Documents\Arduino\libraries\CodeCell\src\CodeCell.cpp:113:31:
error: too few arguments to function 'hw_timer_t* timerBegin(uint8_t,
uint16_t, bool)'
   cctimer = timerBegin(1000000);            /*Set timer frequency to 1Mhz*/
                               ^
In file included from
C:\Users\david\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0\cores\esp32/esp32-hal.h:92,
                 from
C:\Users\david\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0\cores\esp32/Arduino.h:36,
                 from
c:\Users\david\Documents\Arduino\libraries\CodeCell\src\CodeCell.h:4,
                 from
c:\Users\david\Documents\Arduino\libraries\CodeCell\src\CodeCell.cpp:1:
C:\Users\david\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0\cores\esp32/esp32-hal-timer.h:33:14:
note: declared here
 hw_timer_t * timerBegin(uint8_t timer, uint16_t divider, bool countUp);
              ^~~~~~~~~~
c:\Users\david\Documents\Arduino\libraries\CodeCell\src\CodeCell.cpp:114:42:
error: too few arguments to function 'void
timerAttachInterrupt(hw_timer_t*, void (*)(), bool)'
   timerAttachInterrupt(cctimer, &cc_Timer); /*Attach interrupt*/
                                          ^
In file included from
C:\Users\david\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0\cores\esp32/esp32-hal.h:92,
                 from
C:\Users\david\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0\cores\esp32/Arduino.h:36,
                 from
c:\Users\david\Documents\Arduino\libraries\CodeCell\src\CodeCell.h:4,
                 from
c:\Users\david\Documents\Arduino\libraries\CodeCell\src\CodeCell.cpp:1:
C:\Users\david\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0\cores\esp32/esp32-hal-timer.h:40:6:
note: declared here
 void timerAttachInterrupt(hw_timer_t *timer, void (*fn)(void), bool edge);
      ^~~~~~~~~~~~~~~~~~~~
c:\Users\david\Documents\Arduino\libraries\CodeCell\src\CodeCell.cpp:115:3:
error: 'timerAlarm' was not declared in this scope
   timerAlarm(cctimer, 100000, true, 0);     /*Set alarm to trigger every
100ms with auto-reload*/
   ^~~~~~~~~~
c:\Users\david\Documents\Arduino\libraries\CodeCell\src\CodeCell.cpp:115:3:
note: suggested alternative: 'timerStart'
   timerAlarm(cctimer, 100000, true, 0);     /*Set alarm to trigger every
100ms with auto-reload*/
   ^~~~~~~~~~
   timerStart
c:\Users\david\Documents\Arduino\libraries\CodeCell\src\CodeCell.cpp: In
member function 'bool CodeCell::Run(uint8_t)':
c:\Users\david\Documents\Arduino\libraries\CodeCell\src\CodeCell.cpp:413:7:
error: 'timerAlarm' was not declared in this scope
       timerAlarm(cctimer, ((1000000) / _run_frequency_last), true, 0);
/*Set alarm to trigger to new frequency*/
       ^~~~~~~~~~
c:\Users\david\Documents\Arduino\libraries\CodeCell\src\CodeCell.cpp:413:7:
note: suggested alternative: 'timerStart'
       timerAlarm(cctimer, ((1000000) / _run_frequency_last), true, 0);
/*Set alarm to trigger to new frequency*/
       ^~~~~~~~~~
       timerStart
c:\Users\david\Documents\Arduino\libraries\CodeCell\src\CodeCell.cpp: In
member function 'void CodeCell::pinPWM(uint8_t, uint16_t, uint8_t)':
c:\Users\david\Documents\Arduino\libraries\CodeCell\src\CodeCell.cpp:1076:7:
error: 'ledcAttach' was not declared in this scope
       ledcAttach(pin_num, pin_freq, PWM_RES);
       ^~~~~~~~~~
c:\Users\david\Documents\Arduino\libraries\CodeCell\src\CodeCell.cpp:1076:7:
note: suggested alternative: 'ledcAttachPin'
       ledcAttach(pin_num, pin_freq, PWM_RES);
       ^~~~~~~~~~
       ledcAttachPin

exit status 1

Compilation error: exit status 1

This is the codecell repo: https://github.com/microbotsio/CodeCell

Thanks!!!!
David

ricardoquesada, 2025-04-29:
I think it is because Bluepad32 4.1.0 (Arduino IDE) is still based on ESP-IDF Arduino 2.2.x.

And your code is using Arduino Core 3.x

Might take a few months until I update Bluepad32 for Arduino IDE... in the meantime try the template which is updated to latest esp-idf arduino core:

hub.com/ricardoquesada/esp-idf-arduino-bluepad32-template

closed 2025-04-29 by ricardoquesada

da43-hub, 2025-04-29:
Hi,
Thanks.
Wanted ro try the link but i get page not found..

da43-hub, 2025-05-08:
Hi Ricardo,

Thanks for the quick reply and the issue! I appreciate.

Regarding the ESP-IDF Arduino link can you check i get an error the link is
bad..

Thanks again,
David