ricardoquesada/bluepad32 #154: ESP32 C3 SuperMini/Zero and new/more BLE gamepad support

nevsie, 2025-04-16:
I started this in another post, but decided it would be better in here, on its own...
 _Originally posted by @nevsie in [#113](https://github.com/ricardoquesada/bluepad32/issues/113#issuecomment-2776608804)_

First off - ESP32 C3 -  To get information output (within Arduino IDE), be sure to go to Tools -> Use CDC on Boot -> Set to enabled.

Second off, this is specifically looking at BLE connections.

Controller 1 - Terios T3 (Approx £5) - https://www.aliexpress.com/item/32788359483.html
Pair using only "Home + X". 
While Home + A  and Home + Y do use BLE, they are not detected correctly, or data is incompatible.

Controller 2 - 6 In 1 Retro Arcade Console Game Joystick (Approx £10) - https://www.aliexpress.com/item/1005006272959802.html
Pair using "Home + X".
Important note, tap (toggle) the "home" button switches the joystick from analogue to a dpad.
Other pairing does connect, but data is incompatible.

Controller 3 - ShanWan Micro Mini (Q36) (from £4) - https://www.aliexpress.com/item/1005007746811791.html
Pairing is by a single button press, and you use a 3 way switch to select your connections type. Mine was labelled Left X, middle D, right V. The middle position was used. Only Pair once, if you press pair again, the bluetooth address increments and needs a new connection (I am using allow lists!). Press the home button to reconnect after the first time to keep using the same address and connect again.
The Dpad, can be changed to output as a joystick, however, this is not analogue! So not worth it. (press Home + Copy(the two overlapping squares))

I intend to follow up this with a few more of the ShanWan controllers, as they have a nice 8bitdo clone kind of feel to them.

nevsie, 2025-04-22:
[image]
Controller 4 - ShanWan Mini Gamepad Wireless Bluetooth Q37 (£11) - https://www.aliexpress.com/item/1005007585252809.html

Pairing is by a single button press, and you use a 3 way switch to select your connections type. Mine was labelled Left X, middle D, right V. The middle position was used. Only Pair once, if you press pair again, the bluetooth address increments and needs a new connection (I am using allow lists!). Press the home button to reconnect after the first time to keep using the same address and connect again.

nevsie, 2025-04-22:
[image]

Controller 4 - BSP-D11 Stretching Wireless Game Controller (£12) https://www.aliexpress.com/item/1005007439136500.html

Pair by holding X and Home - power button is yellow for this mode.
Yes this is designed for use with a mobile phone, I am using it with a different device (robopad.co.uk) - but it works for BLE so.... its on the list.

nevsie, 2025-04-23:
[image]

Controller 5 - Shanwan Gamepad Q34 - £15 - https://www.aliexpress.com/item/1005008495245614.html

Pair by holding LB + X + Home.  From then on, just hit home button to reconnect.
I would not recommend this controller though, yes it has vibration and hall effect joysticks, but the triggers are not analogue, just one of buttons (Switch style?). I would choose the Terios T3 over this so far.

nevsie, 2025-06-30:
> 
> Did you manage to get any data from the controller? I have exactly the same, they pair, but then no matter which button I press, the bluepad32 reports only zeros. Any hints? Sidenote - the bluepad32 works OK on the same ESP with Wii controller.

Yes, using "Home + X" I got all expected data. 
I experienced what you are saying with the Home + A and Home + Y

If you are just hitting buttons, then the default program that outputs the zeros only shows a minor change in the bitmask section. But all triggers and joysticks worked without issue.  On the example code, the output to watch is: buttons: 0x0004,

One thing you could try - which I have noticed on a couple of controllers. Once connected, if you tap the home button it changes the controller from directional style buttons - digital, to joystick style analogue output. might be worth trying that,

sanchosk, 2025-06-30:
Thanks a lot for the hints.
I tried all pairing modes (Mode+A, X, B, Y). I tried after pairing to switch the mode of operation using the mode button, too.
No matter what I press, I get all zeroes. The controller always pairs no problem, but does not show anything.
I have 3 controllers like that, tested all of them. They all work no problem with Android phone, show all axes and all buttons OK :(
Under no combination would the buttons show 0x0004 - always just pure zeroes :( When I connect old Wii mote, works perfectly OK, reports DPAD and buttons. Shame, as using this controller would be amazing!
Perhaps any other ideas or way how to debug raw messages somehow?
Or another example code?

nevsie, 2025-06-30:
[image]

ESP32 wroom dev board, Terios T3, home+x, you can see it outputting info. Default controller example project.

sanchosk, 2025-07-03:
I cannot find the link to Aliexpress when I bought it. I tried :(
So I guess I'll have to buy a new one - from the supported list...

sanchosk, 2025-07-06:
That's the thing - the controller connects using classic BT, but does not show any values. But no worries, I'll get another one and try that. It's dirt cheap, anyhow...
Thanks a lot for the effort you did here and shared a lot of insights, appreciated!

elik745i, 2025-07-26:
Im trying to connect PS5 to ESP32S3, guess the problem is BLE mode, dont know if PS5 connects using classic or BLE, cant get it to connect to my board, with classic ESP32 no probs, it connects, but not to S3...((

strykeroz, 2025-12-31:
I won't delete the above in case someone else makes my error. I'd thought it couldn't pair using other methods but I was using the allow-list feature which also locked out pairing with X+Home. Erasing the flash allowed me to pair, and now that controller works fine. As with others, I'd fallen into the A+Home trap which only sends the same mostly zeroed data every time, but X+Home is working for me fine now so please ignore the above.

Thanks