ricardoquesada/bluepad32 #155: [Bug]: strange work ble mice

baralgin1003, 2025-04-20:
### What happened?

I bought several new cheap bluetooth mice. They are defined as BLE_5.2. The bluepad cannot connect to them, it keeps giving errors.

```
Mouse 1:
Device found: C3:20:24:01:5E:B3 (public), appearance 0x3c2 / COD 0x580, rssi 205 dBm, name 'BT5.2 Mouse'
Creating device: C3:20:24:01:5E:B3 (idx=0)
BLE scan -> 0
SM_EVENT_IDENTITY_RESOLVING_STARTED
Identity resolved: type 0 address C3:20:24:01:5E:B3
Using con_handle: 0
Bonding information exists for addr type 0, identity addr C3:20:24:01:5E:B3 -> start re-encryption
Connection encrypted: 1
Re-encryption complete, success
Device Information service client connection failed, error=0x6.
Failed to disconnect HIDS client for hids_cid=0, status=2
BLE scan -> 1
--> HCI_EVENT_DISCONNECTION_COMPLETE

Mouse 2:
Device found: 13:08:AC:00:0B:A8 (public), appearance 0x3c2 / COD 0x580, rssi 191 dBm, name 'BT5.2 Mouse'
Creating device: 13:08:AC:00:0B:A8 (idx=0)
BLE scan -> 0
SM_EVENT_IDENTITY_RESOLVING_STARTED
Identity resolving failed for 13:08:AC:00:0B:A8

Using con_handle: 0
SM_EVENT_PAIRING_STARTED
--> HCI_EVENT_DISCONNECTION_COMPLETE
Device 13:08:AC:00:0B:A8 disconnected, deleting it. Reason=0x3e, status=0
Disconnecting device: 13:08:AC:00:0B:A8
Failed to disconnect HIDS client for hids_cid=0, status=2
BLE scan -> 1
connection 13:08:AC:00:0B:A8 already in state 0, ignoring
Deleting device: 13:08:AC:00:0B:A8
BLE scan -> 1
SM_EVENT_PAIRING_COMPLETE: Invalid device for addr 13:08:AC:00:0B:A8
```

At the same time, my other mouse Logi MX MASTER 3S connects perfectly, but there is another problem with it. After rebooting the esp32, the bluepad does not automatically connect to the mouse, you need to turn on the pairing mode on the mouse each time. How can I deal with this?

### Bluepad32 Version

latest from develop branch

### Bluepad32 version custom

4.2.0

### Bluepad32 Platform

Unijoysticle

### Platform version

ESP-IDF 5.4

### Controller

BLE mice

### Microcontroller

ESP32

### Microcontroller board

ESP32-S3-WROOM1 

### OS

Windows

### Relevant log output

```shell

```

### Relevant sketch

```shell

```

ricardoquesada, 2025-04-21:
please add a link to the mouse

baralgin1003, 2025-04-22:
https://waymart.com/product/aoas-ergonomic-bluetooth-wireless-mouse-b-1009/
https://www.aliexpress.com/item/1005006942425344.html
https://www.aliexpress.com/item/1005006821293334.html

baralgin1003, 2025-04-28:
Hi. I have finished the support of BLE devices. Now there is saving and connecting to the last device. If you are interested, you can adjust my code to your style and add it to the main branch. The problem with one mouse has been solved, the second mouse is a Btsteck problem. I wrote about it in the corresponding repository [uni_bt_le.c.txt](https://github.com/user-attachments/files/19942913/uni_bt_le.c.txt)

ricardoquesada, 2025-04-29:
any chance that you can attach the diff or a patch ?

ty

baralgin1003, 2025-04-30:
Hi. Sorry, I have bluepad in my project as a component, it is not linked to version control. I don't know how to make a patch. I will attach all the files where I made changes, please compare them and make the necessary edits. If you have any questions, feel free to ask.

[ptch.zip](https://github.com/user-attachments/files/19978205/ptch.zip)

baralgin1003, 2025-04-30:
A small note on the code. Btstesk added a fix for device compatibility. But this fix is ​​still in the development branch. When the fix is ​​released, please remove the line **gatt_client_mtu_enable_auto_negotiation(0)** from uni_bt_le.с

baralgin1003, 2025-05-03:
The logic of BLE operation turned out to be not quite correct. If in classic BT we connect and maintain the connection until the end, disconnection is considered an emergency situation (left the reception area, device disconnection, etc.), then in BLE disconnection is a shaky operation. For example, after a minute of inactivity, the mouse itself disconnects the connection to save energy. Therefore, you need to constantly alternate scanning between the last connection and the search for new devices. I am currently making some edits to the code and if you are interested, I can send new files.

ricardoquesada, 2025-05-04:
sure. I'm interested.

If you could send a diff, a patch file, or a pull request would be great. ty

trickydee, 2026-10-08:
Hi, 

I believe the PR I just submitted will help with the Logitech MX pairing after a restart issue.

https://github.com/ricardoquesada/bluepad32/pull/236