ricardoquesada/bluepad32 #158: [Bug]: Joy con left and Joy con right cannot be connected at the same time.

LSatan, 2025-05-02:
### What happened?

The Joycons of the Nintendo Switch cannot be connected at the same time.
Individually, however, it works e.g. Joycon Left. If I want to connect Joycon Right now, it comes:

`uni_hid_device_disconnect: Unknown GAP connection type: 0`
`Failed to inquiry name for 50:23:6D:B9:78:44, using a fake one`

### Bluepad32 Version

I don't know

### Bluepad32 version custom

Example: Using Git develop branch commit hash #xxxxxxx

### Bluepad32 Platform

None

### Platform version

E.g: Arduino IDE 2.1.0, or ESP-IDF v4.4.4, or CircuitPython 8.0.0, etc.

### Controller

Xbox Wireless 1708 using firmware 5.15.1923. Link to controller: https://example.com

### Microcontroller

ESP32

### Microcontroller board

ESP32-S3-WROOM1 DevKit from Lolin. This is the store URL: https://example.com

### OS

None

### Relevant log output

```shell

```

### Relevant sketch

```shell

```

LSatan, 2025-05-03:
Generally, multiple devices can be connected. For example, a Bluetooth keyboard, an Xbox One controller, and one of the two Joy-Cons. It seems to me that it doesn't work with devices that have the same vendor ID.

ricardoquesada, 2025-05-04:
You might be triggering this bug: https://gitlab.com/ricardoquesada/bluepad32/-/issues/12

TL;DR: Probably an ESP32 bug.

Do you have any other ESP32 module to try with ?

LSatan, 2025-05-04:
> You might be triggering this bug: https://gitlab.com/ricardoquesada/bluepad32/-/issues/12
> 
> TL;DR: Probably an ESP32 bug.
> 
> Do you have any other ESP32 module to try with ?

I tested it with a node 32 S with the same result. The guys from bug #12 were apparently able to connect both controllers, but the watchdog hung up. What the serial monitor spits out is more similar to the error from: https://github.com/ricardoquesada/bluepad32/issues/135#issuecomment-2576274636
unfortunately I am not familiar with ESP-IDF to test it with it. But I could also order another esp32 to test it. Can you recommend one?
In bug #12 you wrote at the end that you were able to connect 4 joycons without any problems. Did you also try it with Arduino Ide or was it with a different platform?

ricardoquesada, 2025-05-04:
yes, if you try esp-idf, it will be easier to debug it. since we have the logs.

You can try PlatformIO, which is easy easier to use the regular esp-idf.
instructions here:
https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template

LSatan, 2025-05-04:
I'll test the Visual Studio Code example. I've already worked with it. Thanks for the support. Great project. I'll give you feedback on whether it worked.

LSatan, 2025-05-05:
Unfortunately, it's all very frustrating at the moment. I'll try the first example with platform io. I have done everything as described in the instructions. But somehow it keeps complaining that it can't find the main folder for the project. But actually it is there. I wonder what's going wrong. I'll knock everything down tomorrow and redo it. Too bad it doesn't run on the Arduino IDE. Should actually be possible. I wonder if it is due to BTStack or components of the integrated esp lib (with Arduino IDE)?

ricardoquesada, 2025-05-05:
ARduino IDE  should not affect this bug. From my experience, not all ESP32 modules are "created equal".

Try to remove all previous installations of platformio, and the platformio cached files... and start again from step 1. It should work.

LSatan, 2025-05-05:
Yes, it may be that not all esp modules are the same. But if it works with another platform and not with Arduino, then it should be a software and not a hardware problem. At whatever point. But I will try to get it working on the other platform first.

Dendrowen, 2025-10-06:
I have the same issue with the Dualsense controllers
```
20:40:34.921 -> CALLBACK: Controller is connected, index=0
20:40:34.921 -> DConStrol5ler :mode l: DFualSaensei, VIlD=0xe054cd, PI D=0xt0ce6o
20:40:34.921 ->  create virtual device
20:40:46.409 -> uni_hid_device_disconnect: Unknown GAP connection type: 0
20:40:51.009 -> Failed to inquiry name for AC:36:1B:53:48:14, using a fake one
```