ricardoquesada/bluepad32 #163: Add support  for Flydigi controllers

Moddingear, 2025-06-04:
I've got a Flydigi Vader 3 Pro (vid=0xd7d7, pid=0x0041, name='Flydigi VADER3P'), and it doesn't  work, even if I add it as a Steam, Xbox one, Xbox 360 or Android controller in the database. 

I'm not interested in forking over money  to buy it  for you, as I though it would be easy to integrate as it's only  a step for testing a project, but if someone tells me the steps needed to make it compatible with the library, I'll do it.

Moddingear, 2025-06-05:
> yeah... unfortunately I cannot do much without a controller.

Consider that I have all the tools, what are the steps to add a new controller? So that I may add it myself.

ricardoquesada, 2025-06-05:
yes, I should write a doc that explains how to add a controller... in my TODO list