ricardoquesada/bluepad32 #169: Resolving USB HID Conflicts When Using Bluepad32 on ESP32-S3 to Control Your PC with the Right Joystick

AlmacenTecnologico, 2025-07-10:
I love your Bluepad32 project—thank you so much for sharing something so incredible! Based on your experience, could you give me a recommendation? What I’d like to do is use Bluepad32 on my ESP32-S3 to control my PC with the right joystick, but I’m running into USB HID conflicts (it seems the interface is already in use). What would you recommend?

ricardoquesada, 2025-07-10:
You need to provide more details... what kind of "USB HID conflicts" ? 
And what do you mean by "control my PC with the right joystick" ?

AlmacenTecnologico, 2025-07-10:
I am trying to use these libraries that are supposed to be included by default for use with TinyUSB. I have configured the Arduino IDE correctly for my board and enabled the USB HID output for the USB Type-C OTG port.

Libraries:

#include <Bluepad32.h>
#include "USB.h"
#include "USBHIDKeyboard.h"
#include "USBHIDMouse.h"

**And when I compile this, this is what it shows me:**

In file included from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0\libraries\USB\src/USBHID.h:22,
                 from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0\libraries\USB\src/USBHIDKeyboard.h:24,
                 from E:\Bresli\Proyectos electronica\ESP32 S3 como Anfitrion Control XBOX One por Serial\ESP32_S3_Host-Gamepad-Mouse-keyboard-IDE-arduino\ESP32_S3_Host-Gamepad-Mouse-keyboard-IDE-arduino.ino:3:
C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/arduino_tinyusb/tinyusb/src/class/hid/hid.h:87:3: error: 'HID_REPORT_TYPE_INPUT' conflicts with a previous declaration
   HID_REPORT_TYPE_INPUT,      ///< Input
   ^~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/btstack/src/btstack.h:62,
                 from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/bluepad32/include/uni_hid_device.h:8,
                 from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/bluepad32/include/platform/uni_platform.h:15,
                 from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/bluepad32_arduino/include/arduino_platform.h:29,
                 from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/bluepad32_arduino/include/ArduinoControllerData.h:9,
                 from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/bluepad32_arduino/include/ArduinoController.h:13,
                 from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/bluepad32_arduino/include/ArduinoBluepad32.h:12,
                 from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/bluepad32_arduino/include/Bluepad32.h:7,
                 from E:\Bresli\Proyectos electronica\ESP32 S3 como Anfitrion Control XBOX One por Serial\ESP32_S3_Host-Gamepad-Mouse-keyboard-IDE-arduino\ESP32_S3_Host-Gamepad-Mouse-keyboard-IDE-arduino.ino:1:
C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/btstack/src/btstack_hid.h:103:5: note: previous declaration 'hid_report_type_t HID_REPORT_TYPE_INPUT'
     HID_REPORT_TYPE_INPUT,
     ^~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0\libraries\USB\src/USBHID.h:22,
                 from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0\libraries\USB\src/USBHIDKeyboard.h:24,
                 from E:\Bresli\Proyectos electronica\ESP32 S3 como Anfitrion Control XBOX One por Serial\ESP32_S3_Host-Gamepad-Mouse-keyboard-IDE-arduino\ESP32_S3_Host-Gamepad-Mouse-keyboard-IDE-arduino.ino:3:
C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/arduino_tinyusb/tinyusb/src/class/hid/hid.h:88:3: error: 'HID_REPORT_TYPE_OUTPUT' conflicts with a previous declaration
   HID_REPORT_TYPE_OUTPUT,     ///< Output
   ^~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/btstack/src/btstack.h:62,
                 from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/bluepad32/include/uni_hid_device.h:8,
                 from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/bluepad32/include/platform/uni_platform.h:15,
                 from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/bluepad32_arduino/include/arduino_platform.h:29,
                 from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/bluepad32_arduino/include/ArduinoControllerData.h:9,
                 from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/bluepad32_arduino/include/ArduinoController.h:13,
                 from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/bluepad32_arduino/include/ArduinoBluepad32.h:12,
                 from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/bluepad32_arduino/include/Bluepad32.h:7,
[... cut ...]
                 from C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0\libraries\USB\src/USBHIDKeyboard.h:24,
                 from E:\Bresli\Proyectos electronica\ESP32 S3 como Anfitrion Control XBOX One por Serial\ESP32_S3_Host-Gamepad-Mouse-keyboard-IDE-arduino\ESP32_S3_Host-Gamepad-Mouse-keyboard-IDE-arduino.ino:3:
C:\Users\Bresli\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0/tools/sdk/esp32s3/include/arduino_tinyusb/tinyusb/src/class/hid/hid.h:1139:2: error: expected declaration before '}' token
  }
  ^
exit status 1

Compilation error: exit status 1